doudar/SmartSpin2k #431: [BUG] shifter increasing indefinitely (HRM->watts)

ikishk, 2023-01-15:
### Description

HRM->Watts enabled seems to be causing shifter to increase indefinitely, ignoring max watts.  Inclines/shifting does not increase in Zwift, randomly decreases only.

### Steps to reproduce

Load version 22.11.x/22.12.x
reboot a couple times and wait 5-10mins
unit will start increasing resistance by itself... with no paired sensors.

---or---

reboot a couple times until it stops acting weird
pair sensors
start zwift
spin... notice the shifter doesnt increase resistance, only decreases sometimes.

**feature req**
manual "web shifter" limit settings.  While this was happening, the web shifter variable kept increasing.   Having a manual range may help with this, as well as easier to tune limits.  (ie. -5 to 20)

### workaround
disable auto-update
load 22.10.x

ikishk, 2023-01-23:
Been able to narrow this down to a bug introduced in 22.10.7: "Simplified CAD compensation in Powertable by @doudar in #380"  build from 10/6 has been stable so far

doudar, 2023-01-23:
Awesome - Thanks, that will help to track this down.

ikishk, 2023-01-23:
nope, was able to accidently replicate in 10/6.  was connected to zwift, hit manual shifter... then it kept going.

[image]

ikishk, 2023-01-23:
went down to 9/5, rebooted, was still shifting up.  power cycled, still shifting up.   loaded 8/18 (release before the previous power merge) and it stopped.  however the footer still shows 9.5.   is it possible the power table and/or assoc files/cache are not clearing out and surviving between firmware revisions?  off to try and ride again!

doudar, 2023-01-23:
That is a bazar bug. Hopefully I can get some time to replicate it on my end soon.

ikishk, 2023-01-23:
gets weirder.. reflashed 8/18, footer updated.  as soon as i connect zwift... it started up again.  disabled HR->pwr, power cycled. connected to zwift, was fine for 10 mins, then it started again.

eMadman, 2023-01-23:
The frequency of these shifts smells like a possible hardware issue?  Can you try unplugging the shifter - do you still have the frequent shift ups?

ikishk, 2023-01-23:
shifter unplugged half the time, debated that.  doesnt seem to make a diff.

is there a way/build/utility to clear out anything thats possibly cached but not lose basic settings?  seems odd this is happening in 8/18?

eMadman, 2023-01-23:
Could you open up the case and disconnect the shifter from the PCB as well?  It's using a JST connector so it should pop right off - just be sure to pull from the plastic housing and not tug on the wires since the crimps may pull out.  Shouldn't happen but it happens sometimes.

doudar, 2023-01-23:
I just did a quick test with hr->pwr and I couldn't duplicate it. I'll try to do a better/longer test tomorrow.

ikishk, 2023-01-24:
8/18 seemed fine w/ shifter jst disconnected for 15mins in zwift.  power cycled a couple times, used webshifter mid zwift (didnt move as expected).   couldnt reproduce as before.  will load up 10/6 tonight and see how that goes tomorrow. (i like hopping over relevant builds, gives ya a fallback if it doesnt work to test the fallback too)

ikishk, 2023-01-24:
full ride on 10/6, no crazy-upshift issue (jst still disconnected)... no shifting at all (auto or webshifter).  gonna go up to 10/8 and test if that #380 made a diff or not

ikishk, 2023-01-25:
10/8 was stable.  random sim shifting maybe 5 times in an hour, webshifter worked 1/3 the time.  going back up to 10/25.  seems this was a defective 3.5mm plug.  feature req w/ hard webshifter limits might have been a good stopgap if this happens to someone else in the future

ikishk, 2023-01-27:
10/25 definitely works as before (with its little annoyances in mind).  had to be a defective 3.5mm.

ikishk, 2023-02-10:
replaced board.  tried all "major" revisions from 22/9/5 to 23/1/24 and could not replicate.  seems to have been bad board.

closed 2023-02-10 by ikishk

feedthejim, 2023-04-12:
I'm having the same exact issue @ikishk. Did you have to replace the board or just the 3.5mm plug in the end?

eMadman, 2023-04-12:
> I'm having the same exact issue @ikishk. Did you have to replace the board or just the 3.5mm plug in the end?

In @ikishk's case, it was very likely a bridged solder pad for the shifters.  That's the hardest one to diagnose on the V2 but you might have some luck with by using a multimeter for continuity testing.

It can also be the wiring from the 3.5mm plug to the PCB, the shifter cable, or a short in the shifter housing.  

In the case of cable / shifter housing issues, I found I could reproduce it by bending the cable near the housing.  Again, a multimeter may help

I've never actually found a bad 3.5mm plug behaving this way - i've had some faulty ones in terms of tolerances, but never randomly shorting/bridging.

ikishk, 2023-04-12:
old shifters work on the new board tho :) i never reverted to old 3.5mm plug since i already soldered the new one up.  i think it was the board, haven't been able to reproduce since.  many miles put on it thankfully.