doudar/SmartSpin2k #438: [BUG] HW broken? Over temp error and lost bluetooth

xpectnil, 2023-01-23:
First of all and most importantly: thanks to all the people who initiated and contribute to this project.
I've been a silent reader in the past but the smart spin helped me a lot staying sane and fit during the Covid lockdowns. 
You work is highly appreciated. 

### Description

I have a V1 HW version with an Assioma Uno and a Schwinn IC4 and Trainer Day (same behavior on Zwift) on an iPhone 12 Pro and iPad. I built the HW quite some time ago where it seemed to work well. 
The other day, I wanted to use the Smartspin again (my FTP is really down...). I gave the smartspin a reset and updated via USB to the newest Firmware 23.1.22-develop.

I use the suggested settings from the Wiki for the Schwinn IC4.

But I'm getting strange behavior (which may have been the case already before the Firmware update).
Right out of the box, i.e. seconds after booting (and without the motor ever moving) I get an over temp warning.
`[E] (Main): Over temp! Driver is throttling down! ESP32 @ 97.222221 C`

Every component on the PCB has room temperature, I checked. Those warnings pop up every few seconds in the log.
Sometimes, the ERG control loop seems to work; then other times, it doesn't at all. 
Bluetooth is connected, the correct power and RPM is shown in the App, but the smartspin debug screen shows Target Power as 0 while it should be 100.

There seems to be a driver reset and bluetooth is (partially?) lost (it's 20 minutes into a ride):

`[192100][E] (Main): Temperature is now under control. Driver current reset
[1921031(E1 (Main): Scanning from Check BLE Reconnect 0
[192103][E] (BLE Client): Scanning for BLE servers and putting them into a list`

I have Current Watts: 118, Target Watts: 50 in ERG mode - and nothing is happening at all. That's the log of the seconds before. 
I don't quite get it. If the bluetooth connection is lost - how can the power still be displayed in my app and seem correct?

`[1287174][E](FTMS_SERVER): 00 -> Control Request
[1287680][E](BLE_Common): 20 00 4c 00 12 01 3e ba <- 0x1818 | 0x2a63 | CPS[ CD(62.60) PW(76) ]
[1287687][E](BLE_Server): 64 02 8d 00 7c 00 00 00 4c 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(62.00) PW(76) SD(141.00) ]
[1287736][E](Main): Temperature is now under control. Driver current reset.
[1287960][E](ERG_Mode): PowerTable pairs [1][4]
[1287961][E](ERG_Mode): Min Position Set: 145
[1287962][E](ERG_Mode): PowerTable pairs [4][1]
[1287962][E](ERG_Mode): Max Position Set: 19827
[1288194][E](BLE_Server): 64 02 8d 00 7c 00 00 00 4c 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(62.00) PW(76) SD(141.00) ]
[1288696][E](BLE_Common): 20 00 5a 00 13 01 29 be <- 0x1818 | 0x2a63 | CPS[ CD(61.26) PW(90) ]
[1288699][E](BLE_Server): 64 02 a3 00 7a 00 00 00 5a 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(61.00) PW(90) SD(163.00) ]
[1289203][E](BLE_Server): 64 02 a3 00 7a 00 00 00 5a 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(61.00) PW(90) SD(163.00) ]
[1289204][E](FTMS_SERVER): 07 -> Start Training
[1289432][E](Main): Over temp! Driver is throttling down! ESP32 @ 53.333332 C
[1289712][E](BLE_Common): 20 00 76 00 14 01 ff c1 <- 0x1818 | 0x2a63 | CPS[ CD(62.57) PW(118) ]
[1289716][E](BLE_Server): 64 02 db 00 7c 00 00 00 76 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(62.00) PW(118) SD(219.00) ]
[1289717][E](FTMS_SERVER): 00 -> Control Request
[1290228][E](BLE_Server): 64 02 db 00 7c 00 00 00 76 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(62.00) PW(118) SD(219.00) ]
[1290284][E](Main): Over temp! Driver is throttling down! ESP32 @ 96.111115 C
[1290285][E](Main): Scanning from Check BLE Reconnect 0
[1290285][E](BLE_Client): Scanning for BLE servers and putting them into a list...`

I'd like to be more helpful in re-creating the bug. I've programmed quite a bit myself before, so I looked into the code, but it's really not obvious to me what is happening here and why.

Could the hardware / ESP32 / stepper motor drivers be broken? 
Any way how I could test this easily? Then I'd just go ahead and buy new hardware.

### Steps to reproduce

Turn on device, over temp is shown after a second.

That's the log I get seconds after booting:
`[ 26749][E](Main): Over temp! Driver is throttling down! ESP32 @ 88.333336 C
[ 26922][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 27582][E](Main): Temperature is now under control. Driver current reset.
[ 28418][E](Main): Over temp! Driver is throttling down! ESP32 @ 53.333332 C
[ 28423][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 28925][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 29254][E](Main): Over temp! Driver is throttling down! ESP32 @ 88.333336 C
[ 29926][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 30094][E](Main): Over temp! Driver is throttling down! ESP32 @ 88.888885 C
[ 30924][E](Main): Temperature is now under control. Driver current reset.
[ 30927][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 31928][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 32929][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 33930][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 34931][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 35918][E](Main): Over temp! Driver is throttling down! ESP32 @ 53.333332 C
[ 36432][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 36748][E](Main): Temperature is now under control. Driver current reset.
[ 36933][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]
[ 37598][E](Main): Scanning from Check BLE Reconnect 0
[ 37599][E](BLE_Client): Scanning for BLE servers and putting them into a list...
[ 37812][E](BLE_Client): BLE Advertised Device found: Name: F48F92BE899833AEE9, Address: f3:33:07:e8:29:ef, appearance: 0, manufacturer data: a705051001000000000000002200ca, serviceUUID: 0xfe07, txPower: 0
[ 37818][E](BLE_Client): BLE Advertised Device found: Name: yeelight_ms, Address: f8:24:41:c2:19:32, manufacturer data: 6401, serviceUUID: 0xfe87
[ 37843][E](BLE_Client): BLE Advertised Device found: Name: , Address: 54:48:e6:91:4b:ee, manufacturer data: 8f030a109c0b00ec4b91e64854d8
[ 37847][E](BLE_Client): BLE Advertised Device found: Name: F48F92BE891C33AEE9, Address: f3:33:a6:e9:83:4f, appearance: 0, manufacturer data: a705051001000000000000012200ca, serviceUUID: 0xfe07, txPower: 0
[ 37935][E](BLE_Common): 20 00 00 00 05 00 ab 1f <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) ]`

doudar, 2023-01-23:
Chances are that you have a faulty temp sensor in your esp32 dev kit. I've seen a lot of false indications and never a real one yet. But since this isn't an issue with a good working ESP32, I'm hesitant to disable the temperature check for everyone in case it helps somewhere with an actual overheating situation.

For your own purposes, you could comment out the SS2K temperature check in main at:  
https://github.com/doudar/SmartSpin2k/blob/6d38d0fe76222908d80350ce1f96abf6a856a55b/src/Main.cpp#L213
or try increasing the THROTTLE_TEMP in settings.h:
https://github.com/doudar/SmartSpin2k/blob/6d38d0fe76222908d80350ce1f96abf6a856a55b/include/settings.h#L262

I'll do more research and see if this is something we can safely get rid of.

doudar, 2023-01-23:
It shouldn't be causing any problems, but if you aren't using a HR strap, select "none" for your HR device and save. Then SS2K won't periodically scan for a new device as long as your pedals are connected.

xpectnil, 2023-01-23:
Thanks! 

I had HR device set to "none" already. The shifters are set up correctly, I just once more checked with the hint you gave - the debug output and motor direction are correct.

I set power meter to "none" and selected my Assioma Uno once again. The Assioma is fully charged and three feet from the SS2k.

Still - all the time I do get a "Scanning from Check BLE Reconnect 0". 

I don't really know the code... but: isn't there a bug here in the second if?

>   if (((String(userConfig.getConnectedPowerMeter()) == "none") && (spinBLEClient.connectedHR))) {  // Exit if "none" PM and HR is connected
>     bleCheck = 0;
>     return;
>   }
>   if (((String(userConfig.**getConnectedPowerMeter**()) == "none") && (spinBLEClient.connectedPM))) {  // Exit if "none" HR and PM is connected
>     bleCheck = 0;
>     return;
>   }
> 

```
[ 90434][E](Main): Shift -1 pos 0 tgt 0 min -200000000 max 200000000 r_min -2000 r_max 2000
[ 90514][E](BLE_Common): 20 00 2c 00 a4 00 5f de <- 0x1818 | 0x2a63 | CPS[ CD(50.86) PW(44) ]
[ 91515][E](BLE_Common): 20 00 2e 00 a5 00 0b e3 <- 0x1818 | 0x2a63 | CPS[ CD(51.37) PW(46) ]
[ 92517][E](BLE_Common): 20 00 2c 00 a6 00 b1 e7 <- 0x1818 | 0x2a63 | CPS[ CD(51.63) PW(44) ]
[ 93004][E](Main): Shift +1 pos 1 tgt 0 min -200000000 max 200000000 r_min -2000 r_max 2000
[ 93519][E](BLE_Common): 20 00 26 00 a7 00 4e ec <- 0x1818 | 0x2a63 | CPS[ CD(52.02) PW(38) ]
[ 94521][E](BLE_Common): 20 00 20 00 a8 00 ba f0 <- 0x1818 | 0x2a63 | CPS[ CD(54.28) PW(32) ]
[ 95523][E](BLE_Common): 20 00 1e 00 a9 00 22 f5 <- 0x1818 | 0x2a63 | CPS[ CD(54.47) PW(30) ]
[ 95882][E](Main): Scanning from Check BLE Reconnect 0
[ 95883][E](BLE_Client): Scanning for BLE servers and putting them into a list...

```

doudar, 2023-01-23:
> Thanks!
> 
> I had HR device set to "none" already. The shifters are set up correctly, I just once more checked with the hint you gave - the debug output and motor direction are correct.
> 
> I set power meter to "none" and selected my Assioma Uno once again. The Assioma is fully charged and three feet from the SS2k.
> 
> Still - all the time I do get a "Scanning from Check BLE Reconnect 0".
> 
> I don't really know the code... but: isn't there a bug here in the second if? Based on the comment, there should be a check for the HR, shouldn't there? But there's not.
> 
> > if (((String(userConfig.getConnectedPowerMeter()) == "none") && (spinBLEClient.connectedHR))) {  // Exit if "none" PM and HR is connected
> > bleCheck = 0;
> > return;
> > }
> > if (((String(userConfig.**getConnectedPowerMeter**()) == "none") && (spinBLEClient.connectedPM))) {  // Exit if "none" HR and PM is connected
> > bleCheck = 0;
> > return;
> > }
> 
> ```
> [ 90434][E](Main): Shift -1 pos 0 tgt 0 min -200000000 max 200000000 r_min -2000 r_max 2000
> [ 90514][E](BLE_Common): 20 00 2c 00 a4 00 5f de <- 0x1818 | 0x2a63 | CPS[ CD(50.86) PW(44) ]
> [ 91515][E](BLE_Common): 20 00 2e 00 a5 00 0b e3 <- 0x1818 | 0x2a63 | CPS[ CD(51.37) PW(46) ]
> [ 92517][E](BLE_Common): 20 00 2c 00 a6 00 b1 e7 <- 0x1818 | 0x2a63 | CPS[ CD(51.63) PW(44) ]
> [ 93004][E](Main): Shift +1 pos 1 tgt 0 min -200000000 max 200000000 r_min -2000 r_max 2000
> [ 93519][E](BLE_Common): 20 00 26 00 a7 00 4e ec <- 0x1818 | 0x2a63 | CPS[ CD(52.02) PW(38) ]
> [ 94521][E](BLE_Common): 20 00 20 00 a8 00 ba f0 <- 0x1818 | 0x2a63 | CPS[ CD(54.28) PW(32) ]
> [ 95523][E](BLE_Common): 20 00 1e 00 a9 00 22 f5 <- 0x1818 | 0x2a63 | CPS[ CD(54.47) PW(30) ]
> [ 95882][E](Main): Scanning from Check BLE Reconnect 0
> [ 95883][E](BLE_Client): Scanning for BLE servers and putting them into a list...
> ```

Yes, there is a bug there. 
https://github.com/doudar/SmartSpin2k/blob/6d38d0fe76222908d80350ce1f96abf6a856a55b/src/Main.cpp#L565-L577

On line 565, one of those items should be userConfig.getConnectedHeartMonitor()
as well as on line 573

Do you want to make a PR so you get credit for it, or should I make the changes quickly?

xpectnil, 2023-01-23:
Go for it. :-) 

Still not working as expected, though... 

At ~ 118 W of current power and a target power of 48 Watts - nothing is happening for a minute. It doesn't shift the resistance down. 
Any idea what the other problem could be? 

```
[381242][E](BLE_Server): 64 02 20 01 a6 00 00 00 74 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(83.00) PW(116) SD(288.00) ]
[381242][E](FTMS_SERVER): 07 -> Start Training
[381755][E](BLE_Server): 64 02 20 01 a6 00 00 00 74 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(83.00) PW(116) SD(288.00) ]
[381756][E](FTMS_SERVER): 00 -> Control Request
[382264][E](BLE_Common): 20 00 76 00 a4 01 44 54 <- 0x1818 | 0x2a63 | CPS[ CD(85.57) PW(118) ]
[382267][E](BLE_Server): 64 02 2c 01 aa 00 00 00 76 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(85.00) PW(118) SD(300.00) ]
[382776][E](BLE_Server): 64 02 2c 01 aa 00 00 00 76 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(85.00) PW(118) SD(300.00) ]
[383278][E](BLE_Common): 20 00 72 00 a5 01 0f 57 <- 0x1818 | 0x2a63 | CPS[ CD(85.93) PW(114) ]
[383284][E](BLE_Server): 64 02 21 01 aa 00 00 00 72 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(85.00) PW(114) SD(289.00) ]
[383285][E](FTMS_SERVER): 07 -> Start Training
[383672][E](PowTab): |    0|   57|  105|  134|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|
[383675][E](PowTab): |    0|-1256|  256| 1200|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|
[383796][E](BLE_Server): 64 02 21 01 aa 00 00 00 72 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(85.00) PW(114) SD(289.00) ]
[383797][E](FTMS_SERVER): 00 -> Control Request
[384305][E](BLE_Common): 20 00 5c 00 a7 01 9e 5c <- 0x1818 | 0x2a63 | CPS[ CD(86.35) PW(92) ]
[384313][E](BLE_Server): 64 02 ed 00 ac 00 00 00 5c 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(86.00) PW(92) SD(237.00) ]
[384817][E](BLE_Server): 64 02 ed 00 ac 00 00 00 5c 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(86.00) PW(92) SD(237.00) ]
```

doudar, 2023-01-23:
If you could send a full log that would be great. 
The best is if you can use Logbert https://github.com/couchcoding/Logbert/ .
Turn on the UDP logs in the settings.
Then in logbert (connected to the same network as SS2K), do Custom UDP reciever, port 10000. 
Then when you're done collecting logs, put them in pastebin and post the link here. 

I'll need a short ride with a few different ERG targets. Note that the ERG minimum is the min brake watts as well.

xpectnil, 2023-01-23:
Thanks for your support.

Logbert seems to be a .NET program and I've got a Mac without .NET installed.
I used the Flasher Tool and did a full log - I hope that works.

Here's 300kB of log file:
https://pastebin.com/0DF0ehpd

doudar, 2023-01-23:
> Thanks for your support.
> 
> 
> 
> Logbert seems to be a .NET program and I've got a Mac without .NET installed.
> 
> I used the Flasher Tool and did a full log - I hope that works.
> 
> 
> 
> Here's 300kB of log file:
> 
> https://pastebin.com/0DF0ehpd
> 
> 
> 
> Let me know if there are any other tests I could run.
> 
> 

Definitely looks like an app issue. What app is it that you're using so I can test and fix?

xpectnil, 2023-01-23:
Confirmed - it's an app issue. I just tried Zwift - works like a charm.
At least I could fix a bug along the way and contribute back to the project. 

I'm had tried out TrainerDay. https://trainerday.com/  The only reason for it is that it's free in its basic version.

Any other recommendation for an App with ERG mode that works, doesn't cost that much?
I want to run some 30-60 min FTP builder track and watch videos along the way (ideally iPhone for running the ERG App so I can use my full iPad for watching videos).

Any recommendation that works really well?  No need for fancy videos or real-world simulations.

doudar, 2023-01-23:
I know @eMadman uses MyWoosh. 
I'll look at trainerday and see what the issue is. I'm sure I can fix it.

eMadman, 2023-01-23:
Mywhoosh is great - completely free at this time and similarly feature rich to Zwift.  Haven't played with it in Erg but I used to do a lot of on-desk testing with it for sim mode

doudar, 2023-01-23:
If you want super simple and no install, this web app is pretty cool for erg training. I just tested it and it works. 
https://flux-web.vercel.app/
https://github.com/dvmarinoff/Flux

xpectnil, 2023-01-24:
Thanks for your suggestions. I'll check them out.

xpectnil, 2023-01-27:
> Mywhoosh is great - completely free at this time and similarly feature rich to Zwift. Haven't played with it in Erg but I used to do a lot of on-desk testing with it for sim mode

Thanks, I tried it today. It's really a clone of Zwift and - unlike the App Store Reviews - ran very stable.
SS2k worked perfectly well with Mywhoosh, no problem whatsoever.

xpectnil, 2023-01-27:
One bug got fixed, the other strange behavior only happens with Trainer Day - so it seems to be an app issue, not a SS2k issue.

closed 2023-01-27 by xpectnil