doudar/SmartSpin2k #504: Resistance Increases Indefinitely[BUG] (short description)

RyzinEnagy, 2023-12-10:
### Description

As soon as I start a ride on Peloton, the resistance slowly begins increasing indefinitely and attempts to keep increasing it past 100 percent. The shifter buttons do nothing when this is happening. When I leave the ride, it resets back to where it was before the ride. 

**Expected behavior**
Resistance to adjust properly

**Actual behavior**
As soon as I start a ride on Peloton, the resistance slowly begins increasing indefinitely and attempts to keep increasing it past 100 percent. The shifter buttons do nothing when this is happening. When I leave the ride, it resets back to where it was before the ride.

doudar, 2023-12-10:
It sounds like possibly it isn't receiving resistance information from the bike (if this is a Peloton bike.) 
Are you able to post any logs?
Is this using the Peloton App or Peloton Bike, or both?

RyzinEnagy, 2023-12-11:
It's on the Peloton Bike, and I'm beginning a ride on its own tablet through the Just Ride option (though it happens when I open any ride on the Peloton). 

[Here is a log I took](https://pastebin.com/a73vF5NN).

eMadman, 2023-12-11:
Can I ask you to enable BLE RX logging as well?  It'll give much more information in the log:
[image]

RyzinEnagy, 2023-12-11:
Sure. [Here is a new log](https://pastebin.com/qeXdWUzJ).

eMadman, 2023-12-11:
The -648 value for resistance is odd.

Can you send a me a picture of the sticker on the back of your bike's tablet as well as a picture of your about tablet screen?  You can access the screen from the 3 dots menu on the bottom right of the Peloton Tablet - I think it's labeled About Device or something.  Mostly curious if you've gotten an update that breaks something

Feel free to crop out or blur out your serial.  I'm most interested in the Bike's Model Number, the Software version, and the last updated date.  
[image]
[image]

RyzinEnagy, 2023-12-11:
[image]
[image]

Here you go.

doudar, 2023-12-11:
I've identified the issue. Should have a fix in an hour or so. For some reason, your bike is the first one I've run into that sends "4a" as the resistance code instead of "49".

RyzinEnagy, 2023-12-11:
Great, much appreciated.

doudar, 2023-12-11:
Okay, this is because 4a is a raw resistance value, not the calibrated resistance value the we can get most of the time. 

Try using the SmartSpin2k in headless mode (tx mode) and see if you still have the same issue. 

Regardless, the stepper shouldn't run away, so that should be fixed in the attached firmware. Developer tools->upload firmware->admin/admin. Extract it (I can't upload binaries), and make sure it's named exactly firmware.bin . Everything will work like before, it's just not going to use the resistance off of the bike to determine limits. It's a from a newish branch I've been working on - 90% stable, please let me know if you notice any reboot issues (I think I fixed them earlier tonight.)  

It's going to be a little while before I can work on the ultimate fix for this, which is downloading the resistance calibration chart off of your bike and implementing it, but it still needs to be done, hopefully within a week. I'll need @RyzinEnagy for testing it. 
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/13631382/firmware.zip)

RyzinEnagy, 2023-12-11:
In headless mode, the issue occurs at all times when the bike is on, regardless of whether a ride is in progress.

This new firmware worked. I was able to do a workout with ERG as normal.

doudar, 2023-12-24:
Fixed by #498

closed 2023-12-24 by doudar

cjwhitedev, 2024-01-15:
@doudar - i am having this issue as well. i have updated to [23.12.29](https://github.com/doudar/SmartSpin2k/releases/tag/23.12.29), though that has not resolved the problem.

doudar, 2024-01-15:
@cjwhitedev , please post a log with the "BLE RX logging" setting enabled and I'll take a look. 

Preferably with these directions, but if you can only do web logging, that's okay as well: 
https://github.com/doudar/SmartSpin2k/wiki/Viewing-logs-via-UDP 

Thanks for the report!

cjwhitedev, 2024-01-20:
I'm sure this is evident in the log, but after a reboot, and once confirming the logger was working, i hit shift up one time on the web shifter, then it ran away on its own.

[Custom UDP Receiver (Port 10000).txt](https://github.com/doudar/SmartSpin2k/files/13998544/Custom.UDP.Receiver.Port.10000.txt)

doudar, 2024-01-20:
Whoa. Thats a lot of shifts. I'll check into it - thank you.

cjwhitedev, 2024-01-22:
that might be the case, but it happens when i connect to zwift as well. just something to keep in mind. whatever gets it started, it just keeps going and going.

cjwhitedev, 2024-02-04:
@doudar should i open a new ticket for this? or should this ticket be re-opened?

doudar, 2024-02-04:
Sure, go ahead. Yours is different from this closed issue. I'll need OS and other circumstances as well. I haven't been able to reproduce it.