doudar/SmartSpin2k #510: Loss of power in sprints.

serbasan, 2023-12-28:
### Description

When I do stable training in ERG mode with power variations everything works perfectly.
When I have done a race without ERG and without SIM (I disable resistance in devices) in Zwift, using only the switches to change gears everything has worked well with powers no higher than 300W.
I have the problem when doing a sprint, going up gears with the switch reaching 400W-600W, it always drops the gears suddenly and I lose all my power. It is awful.

SmartSpin2k has very little resistance and I can no longer shift gears with the switch.

The logs say "Shift Blocked by stepper limits" and I can't shift up. I can only go down and continue going down beyond the limit many times (I hear how I reach the lower limit and the motor loses steps). After that I can go back up to the upper limit.

If I go up gears again and do a powerful Sprint, the same thing happens again.

I have also tried in SIM mode and the same thing happens.

It happened to me with the previous firmware and the new one that works with the torque table.

I have Max Bike Brake Watts set to the maximum

### Steps to reproduce

1) Start a route in SIM mode or with resistance disconnected.
2) Use the switches to up and down gears.
3) After a few minutes, go up the gears with the switch and do a 500W sprint.
4) The resistance will drop suddenly and you can no longer increase resistance with the switch.
5) Lower resistance beyond the limit many times.
6) You can now increase resistance to the limit.

**Expected behavior**
Use the switches to go up and down gears and be able to do a high-watt sprint without suddenly losing resistance.

**Actual behavior**
When I do a route in SIM mode or without resistance control connected, if I do a sprint I lose all resistance. If I try to shift up I can't and in the logs it says "Shift Blocked by stepper limits"

### Additional Information
I am using a hardware adaptation for the Keiser M3 as I have already mentioned in the [discussion section](https://github.com/doudar/SmartSpin2k/discussions/494)

doudar, 2023-12-28:
You did a fantastic job collecting logs for this - Thanks!

Could you send one log from the current develop firmware with Torque Table? That would help me determine the cause much easier since there were some significant changes made to the upper boundary setting.

doudar, 2023-12-28:
Also...Looking at this more - is there any chance that the stepper motor is losing steps (slipping) when you go to higher resistance?

serbasan, 2023-12-28:
Hello.
I doubt it will lose steps since it does not require much torque to move the system, it runs very smoothly and has a very short path. I remember that I modified "driver.microsteps(8); // Set microsteps to 1/8th" to 8 steps in main.cpp
I think the problem is that my magnet system has a spring to always bring the magnet to the position of least resistance (gear 1).

There must be a moment when the firmware leaves the motor without power, perhaps during the ERG changes and the spring always pushes towards gear 1 causing a false loss of steps.

Is there a way to always keep the engine locked when changing direction?

I think this has never been a problem on the bicycles in which SmartSpin2k has been designed but on the Keizer 3M it has that difference with the spring that always pushes to gear 1

If there are any changes I can make to the firmware, compiling and testing it is no problem on my part.

How do I get a log of the firmware I have loaded?

Thank you.

doudar, 2023-12-28:
> Hello. I doubt it will lose steps since it does not require much torque to move the system, it runs very smoothly and has a very short path. I remember that I modified "driver.microsteps(8); // Set [...]

Set both of these to false:
https://github.com/doudar/SmartSpin2k/blob/7f2484e092204b50774cf323e30794fc34ab7fe2/src/Main.cpp#L376-L381

Remove the *.5 here:
https://github.com/doudar/SmartSpin2k/blob/7f2484e092204b50774cf323e30794fc34ab7fe2/src/Main.cpp#L452-L453

Be careful! The stepper driver might run hot after these modifications!

doudar, 2023-12-28:
As far as determining what version it's running, you can see it on boot up in the very beginning of the logs, or at the bottom of index.html. 

I was able to tell it wasn't the Torque Table firmware because it's the older power table being displayed in the logs.

serbasan, 2023-12-28:
I have made the changes. I'll try as soon as possible and comment. The branch I have selected is that:

[image]

serbasan, 2023-12-29:
I've been testing and the loss of power when sprinting happens again.

I have removed the shift spring from the Keizer and changed the belt in a continuous loop. In this way I have canceled the force of the spring.

I have done several springs again and the problem happens but since the spring is no longer there I do not lose all the power, I simply cannot increase more resistance with the switch.
At that moment I realized that the motor loses steps so I already know what is happening:

When you do a sprint the power goes up a lot 400W-600W. At that moment the Keizer magnet exerts a lot of magnetic force and must be harder and the motor loses steps because it does not have the strength to move it.

In short, it is not a firmware error. It is due to the hardware configuration with the Keizer M3. When a lot of power is exerted (400W+) the magnet becomes very hard and the motor cannot move and loses steps. At that moment the motor is disconnected and the spring moves the magnet to the position of least resistance (gear 1).

Would it be possible for the motor to not disengage when it loses steps?

Removing the spring largely solves the problem but prevents you from being able to use the bike manually.

doudar, 2023-12-30:
Maybe reduce the size of your drive pulley to just bigger than the bolt and then put the nut at the end of the pulley instead of recessed inside it?

Or you could add another linkage to the belt (another lever) to increase the mechanical advantage. 

Or, you could remix the entire SS2K case and change the gear ratio there.

serbasan, 2023-12-30:
I'll think about how to improve the transmission strength. I'm modifying the Fusion 360 box and it could do something.
With firmware modifications and adding a fan in the case it improves somewhat.
For ERG and SIM training the system works very well.

By the way, it's beside the point but since I installed the latest firmware, OTA updates no longer work for me. I have to do them by cable. The modified firmware is uploaded, it does the entire process but maintains the previous one.

For my part, this issue can be closed. It's not a firmware problem. It is a hardware problem with the Keiser M3.
When I get the modification of the tested box I will comment on it.

Thank you very much for the help.

serbasan, 2024-01-01:
Hello and Happy New Year.

I have made a hardware modification for the adaptation with Keizer.
I've gone from this:

[image]

To this:

[image]
[image]
[image]
[image]

Much more torque has been achieved and now I can do a sprint and go up gears without problems until I can't go any further.

I will continue training and adjusting the STL of the box and adapter until I am convinced that everything fits.
With each change I get it to work better. I'm very excited.

Can I make STLs public on this Github project and on platforms like Thingiverse?

doudar, 2024-01-01:
Awesome! That was a fast fix!

Yes, if you'd like, you can contribute the hardware mod back here. I'd love to add it to the repo so others can use it if they wish. I don't have any issues with thingaverse either. 

Happy new year!

eMadman, 2024-01-01:
> Awesome! That was a fast fix!
> 
> Yes, if you'd like, you can contribute the hardware mod back here. I'd love to add it to the repo so others can use it if they wish. I don't have any issues with thingaverse either. 
> 
> Happy new year! 

Awesome that you have it working! 

Just ensure you're using gpl version 2 when you post it to thingiverse or others!

It would also be awesome to have it in the repo here as a mod!

doudar, 2024-03-23:
Curious if you we able to solve this. There have been a few people asking on the Facebook group about the M3

serbasan, 2024-03-23:
Hello.
With the modifications I made to the SmartSpinv2 box and the top torque adapter for the Keizer M3 gears everything works perfectly. I've been using it for several months now and I had already forgotten all the time I spent adapting everything to the Keizer M3. When I do my workouts I turn it on, use it and everything works perfectly. I really enjoy how well it works.

The only thing you have to keep in mind is that I use SmartSpin2k together with [another project](https://github.com/ajs123/KBikeBLE) to be able to get power and cadence via Bluetooth, so it may be a little more complex for Keizer M3 users, but it works so well that I recommend the effort.

Maybe it can only be used with heart rate as discussed here.

The files I have designed [are these](https://www.thingiverse.com/thing:6545641).

This modification is part of version 2 since I have assembled it myself by purchasing the v2 board. The box is attached with the version of the inserts for the screws.
There is a version to mount a fan.
Everything is secured with a front flange and the two 3d fasteners. Everything is well secured like this.

You can take all the files and add them to your repository if you wish and if you have questions or want me to make any modifications, you can tell me.

Thank you very much for everything and it would be a great honor if someone can take advantage of it for their Keizer M3.

Greetings.

doudar, 2024-03-23:
That's pretty awesome! Any chance you have Facebook or could make an intructable (https://www.instructables.com/) for it? I know a lot of Keiser users are interested.

closed 2025-01-24 by doudar