doudar/SmartSpin2k #514: [FR] Bowflex IC SE Bike support

Codelica, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
A problem of my own making. :) I have a new Bowflex IC SE bike and have been working with @eMadman to outfit the bike with a SmartSpin2K.  Hardware-wise I'm setup with a prototype that fits well, but have run into an issue streaming data (cadence, power, etc) from the bike. 

**Describe the solution you'd like**
To work toward supporting the [Bowflex IC SE bike](https://www.bowflex.com/exercise-bikes/ic-bike-se/101012.html). 

**Describe alternatives you've considered**
Just giving up. ;) 

**Additional context**
So currently I have a SmartSpin2k attached, connected to wifi, etc.  On startup it also seems to connect to the bike, as I see "App Connect" on the Bowflex screen (and it isn't from another app).  However in Zwift using the SSK2 connection (or the SSk2 Web dev screen) data doesn't stream. (always 0 cadence, 0 power).  

@eMadman asked if I'd take shots from nRF Connect for Fitness machine (0x1826) and Cycling Power (0x1818).  There is Fitness machine (0x1826) but I don't see Cycling Power (0x1818). You'll see there is Cycling Speed and Cadence (0x1826) and another Unknown Service.  A bunch of screenshots attached to start.

Ask way if there are further questions.  No rush on this, just hoping to get it working "someday". :) 

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]

doudar, 2024-01-15:
This might not be a hard thing to fix. I think what could be going on is that the first service that's advertised is the Cycling Speed and Cadence service (which we don't use), so it's probably getting overlooked as a useful device. 

Is there any useful information showing up on the logs when you first power on the SmartSpin2k and it tries to scan for it?

Codelica, 2024-01-15:
In case it matters, this is the connection long from having the SS2K on and sitting at the Dev tools debugging screen, then turning on the bike on (after it's been off a while):

```
[1085488][E](BLE_Client): Matching Device Name: ICSE
[1085489][E](BLE_Client): PM String Matched Any
[1085489][E](BLE_Client): Failed to set service!
[1085489][E](BLE_Client): doConnect set on device: 0
[1085774][E](BLE_Client): Initiating Server Connection
[1085775][E](BLE_Client): trying to connect to Fitness Machine Service
[1085775][E](BLE_Client): Forming a connection to:  04:cd:15:01:38:90
[1085778][E](BLE_Client):  - Created client
[1086276][E](BLE_Client): Connected to: 04:cd:15:01:38:90 RSSI -49
[1087126][E](BLE_Client): Successful 0x2ad2 subscription.
[1087126][E](BLE_Client): Registered PM on Connect
[1087126][E](BLE_Client): Device Connected
[1087127][E](BLE_Client): We are now connected to the BLE Server.
[1114537][E](ERG_Mode): Using Resistance Travel Limits
[1150269][E](ERG_Mode): Using Resistance Travel Limits
[1185991][E](ERG_Mode): Using Resistance Travel Limits
...
```

Codelica, 2024-01-15:
This may be more interesting.  Doing the same startup as above, once the bike connects I actually *can* see changing power and cadence info in the SS2k developer screen.   However, once I launch Zwift and it connects to the SS2k, the power and cadence info in the developer screen no longer works (stuck to 0) and Zwift shows 0's for everything.   Log is attached.

[SmartSpin2-log.txt](https://github.com/doudar/SmartSpin2k/files/13934764/SmartSpin2-log.txt)

doudar, 2024-01-15:
Also, on the next log, if you could check "enable BLE RX Logging" that would be great. It's a dumb setting anyway :)

Codelica, 2024-01-15:
Thanks, I'll give a shot tonight.  👍

Codelica, 2024-01-16:
So i set BLE RX Logging on and saved, then applied the firmware.bin and rebooted.  The version still showed `23.12.29-develop` which is hopefully expected.  

Then:

- Starting with the SS2k on
- Powered up the bike and it connected
- Pedaled briefly to make sure data streamed on the Web developer page
- Started up Zwift and connected to the SS2k
- Confirmed data seems to have stopped streaming on the Zwift connection setup page
- Start an acivity in Zwift to confirm there stlll wasn't data
- Quit Zwift

Log attached.

[ss2k-log-1-15-24.txt](https://github.com/doudar/SmartSpin2k/files/13945103/ss2k-log-1-15-24.txt)

eMadman, 2024-01-16:
it looks like it reverted itself to the current release version.  Can you
disable firmware automatic updates in the settings and try uploading the
new firmware.bin again?

Codelica, 2024-01-16:
Ooops, missed that setting.  Will give it another shot tonight.

Codelica, 2024-01-17:
So I disabled Automatic Updates and rebooted to make sure it was still off.  Then updated the firmware and rebooted.  Still showed the same `23.12.29-develop` version so I updated the firmware one more time and rebooted, but still saw that version again.  Both times said it updated fine, so I assumed it's been changed. (?)

Did the same steps as above except I didn't stop peddling when launching Zwift, which was sort of interesting in that it basically locked the power + cadence at some level I was at while Zwift connected.  From that point on it just stuck to that data even without me physically peddling at all.  Then I quit Zwift. 

[ss2k-log-1-16-24.txt](https://github.com/doudar/SmartSpin2k/files/13958097/ss2k-log-1-16-24.txt)

doudar, 2024-01-20:
> So I disabled Automatic Updates and rebooted to make sure it was still off. Then updated the firmware and rebooted. Still showed the same `23.12.29-develop` version so I updated the firmware one more time and rebooted, but still saw that version again. Both times said it updated fine, so I assumed it's been changed. (?)
> 
> Did the same steps as above except I didn't stop pedaling when launching Zwift, which was sort of interesting in that it basically locked the power + cadence at some level I was at while Zwift connected. From that point on it just stuck to that data even without me physically peddling at all. Then I quit Zwift.
> 
> [ss2k-log-1-16-24.txt](https://github.com/doudar/SmartSpin2k/files/13958097/ss2k-log-1-16-24.txt)

Thanks! Just seeing this - didn't get a notification for some reason

doudar, 2024-01-20:
@Codelica #518 should finally fix this. 
Here's a binary. Please let me know.
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/13998065/firmware.zip)
Have a great day!

Codelica, 2024-01-21:
Thanks!  I just gave it a quick try an it seems to be working fine. 👍  Thanks for the speedy work on this. I'm looking forward to a real ride once I've recovered from a weekend of shoveling and skiing. ⛄

doudar, 2024-03-23:
fixed by #518

closed 2024-03-23 by doudar