doudar/SmartSpin2k #515: Not working with TrainerDay

docholiday1234, 2024-01-19:
SS2K is not working with Trainer Day app iOS 17.0.3

### Steps to reproduce

1. Open App
2. Connect SS2K
3. Start workout
4. App take control of SS2K and raises or lowers resistance until it bottoms out. I

**Expected behavior**
It should raise or lower resistance consistent with ERG mode wattage setting

**Actual behavior**
resistance seems to acting sporadically

### Additional Information
[Custom UDP Receiver (Port 10000).txt](https://github.com/doudar/SmartSpin2k/files/13994681/Custom.UDP.Receiver.Port.10000.txt)

doudar, 2024-01-20:
Thank you for the logs!!

The issue appears to be that TrainerDay sends ERG targets at least once per second. Currently the SmartSpin2k ERG algorithm uses the lookup table for the first two seconds after it receives a ERG target and then after the two seconds elapses, it uses a normal positive feedback control loop. With these rapid fire ERG targets, the current scheme never gets a chance to go into a positive control mode and the rapid ERG targets keep it spiraling out of control.

doudar, 2024-01-20:
New branch #516

doudar, 2024-01-20:
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/13996008/firmware.zip)
@docholiday1234 , give this firmware a try. 

Instructions: 
https://github.com/doudar/SmartSpin2k/wiki/Loading-Software
Use the "Updating your existing software" section, except use the binary from the zip file attached to this message.

docholiday1234, 2024-01-20:
Hi, It shifted back and forth, immediately and constantly increasing and decreasing the resistance. Then I turned up the watts and it seems to increase resistance but then when I turned back down the watts it still increased resistance although it never 'bottomed out' and now that I'm not even pedaling it is still changing resistance. 

[Trainer Day Log 2.txt](https://github.com/doudar/SmartSpin2k/files/13996132/Trainer.Day.Log.2.txt)

doudar, 2024-01-20:
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/13996481/firmware.zip)
This is the latest build. It should work for you.

closed 2024-02-25 by doudar