doudar/SmartSpin2k #523: [BUG] Uncontrollable Upward Shifting

cjwhitedev, 2024-02-08:
### Description

Whenever I try to use my smartspin it shifts uncontrollably upward.

### Steps to reproduce

1. Turn on unit
2. Shift the unit from any source (the web interface or via swift)
3. It will now shift updward indefinitely.

Please reference my comments in #504.

[Custom UDP Receiver (Port 10000).txt](https://github.com/doudar/SmartSpin2k/files/14211589/Custom.UDP.Receiver.Port.10000.txt)

cjwhitedev, 2024-02-08:
What do you mean by disconnected? If it is not attached to zwift?

doudar, 2024-02-08:
Physically disconnecting the shifter buttons. I'm thinking this is some type of hardware issue or possibly the shifter wires inside the case are in close proximity to the stepper wires (like caught together in the case seam)

cjwhitedev, 2024-02-08:
Its been unplugged for a couple weeks. I just plugged in the power, nothing else, and it resumed climbing indefinitely. I let it run for a few seconds before unplugging it again. Here is that log.
[Custom UDP Receiver 2 (Port 10000).txt](https://github.com/doudar/SmartSpin2k/files/14212081/Custom.UDP.Receiver.2.Port.10000.txt)

doudar, 2024-02-08:
That was shifting without any input from you? Gotta be some type of hardware issue.

cjwhitedev, 2024-02-08:
I have the latest release installed. This is where i bought it from:
https://www.etsy.com/listing/1356903775/smartspin2k-v3-spin-bike-smart-trainer

eMadman, 2024-02-08:
Just want to confirm that the shifter and its cable were disconnected from the breakout cable during this test - I want to make sure it's not a hardware issue relating to the shifter before I ask you to open up the unit.  

There's a small possibility that the motor's cable is twisted around the internal wiring causing some interference

cjwhitedev, 2024-02-09:
The only thing plugged in was the power. I will open the unit up later and report back.

eMadman, 2024-02-09:
> The only thing plugged in was the power. I will open the unit up later and report back.

Sounds good.  We have a PDF manual for assembly/disassembly here: https://www.smartspin2k.com/diy-building-instructions - the most fiddly part is routing the cable internally but it should be pretty straightforward

cjwhitedev, 2024-02-09:
Nothing in here looks tangled or twisted in here to me.

[image]

eMadman, 2024-02-11:
Ahh!  it's a V2!  My best guess, this might be an issue with the 3.5mm headphone connector or a failure on the PCB.

Can you email me at emadmanmakes@gmail.com?  I think we can work something out since you bought this from me last year.

doudar, 2024-02-24:
Did we ever figure out what the (hardware?) issue was here?

eMadman, 2024-02-25:
@doudar - the v3 upgrade kit sorted the issue out.  Pretty sure it's something wrong with the old circuit board.  The pic looks like it was from my last batch of esp32s before the transition when I had to start soldering that offshoot capacitor into place or they wouldn't accept firmware properly.

doudar, 2024-02-25:
Okay! Thx closing this then.

closed 2024-02-25 by doudar