doudar/SmartSpin2k #527: How to get into Smartspin2k settings

wassa22, 2024-02-19:
Click on the hyperlink but it doesnt get into settings. Can see the unit on the network. Have searched around but not sure what I need to do.

Cheers.

Warwick.

wassa22, 2024-02-19:
Ok manged to get in to settings....error was not knowing  password. Changed settings which I thought was turning off bluetooth and when I hit save it disconnected and now while I can still see and connect to it I cannot get back into settings. Please advise what I need to do

eMadman, 2024-02-19:
What's your goal using it with mywhoosh and QZ concurrently?  I'd suggest saving the QZ integration to ride with Peloton.  Adding it into the mix for use with MyWhoosh only adds complexity that will make it harder to troubleshoot.  

To ride with Mywhoosh:

- Turn on the SmartSpin2K
- Turn on your bike and HR monitor
- launch the app and look for SmartSpin2K in the bluetooth pairing screens - select it for your HRM, power meter, controllable smart trainer and cadence sensor
- Ride.

wassa22, 2024-02-19:
I have the yesoul bike which requires qz to get the data from the bike.... my goal is to be able to have the auto resistance work in free rides in mywhoosh and possibly zwift. Not interested in Peleton.

doudar, 2024-02-20:
Thanks @wassa22 ! If you're all set, I'll close this issue tomorrow. In the meantime, does the Yesoul not output regular FTMS? (SmartSpin2k nor Zwift sees it?)

Someday we should make a matrix of bike compatibility. There's so little time in the day.....

wassa22, 2024-02-20:
No the version I have doesn't which is why I need to intergrate qz into the mix. Had to load an app on to the bike to send the data.

App looking at the log Roberto advised that qz app couldnt find SS2K so that is the issue now to figure out. The initial issue of getting into the web page is all good.

doudar, 2024-02-20:
Okay! Sounds good. All I know about the QZ integration is that you need to have "none" selected for all 3 BLE devices on the SmartSpin2k. QZ controls the SmartSpin2k directly via a special custom characteristic we developed together. All that configuration can be done in Roberto's app. Either him or @marklogan17 are the experts in that area.

wassa22, 2024-02-20:
Cheers.... now cant back into it to check settings... how long should it take to connect to the Network. More headaches than I expected..🤣

wassa22, 2024-02-20:
Could be why qz couldnt see it....I seem to really struggle to connect to it. .

marklogan17, 2024-02-20:
OK so where are you right now?  You can get into the SS2K web page and see the settings but you can't connect to QZ?
1) With QZ off can you use the physical buttons to change resistance up and down?
2) In SS2K settings have you turned off all the bluetooth?
3) In QZ in the settings under Smart Spin there is a drop down, when you drop that down do you see the SmartSpin?
4) If so did you select it in the drop down AND hit OK.  If yes, then completely exit the app - kill it - go back in and make sure you see the Smart Spin configured there.
5) If you made it this far power off the bike, power of SS2K and kill QZ.  Now power on the bike.  Power on SS2K - and wait until the physical buttons move the resistance up and down.  Start QZ.  After it connects to the bike go to the resistance tile and hit the + or the -.  Did it work?  If so go to the gear tile and hit + or -.  did it work?

Let us know where you fail above and that will help us diagnose

wassa22, 2024-02-20:
Ok I couldnt get back into web settings for SS2K so rebooted it back and went and set network up again. Once I did that and rebooted it I cannot get back into it to turn bluetooth off as I get the error saying the SS2K is unreachable same as yesterdY.

In QZ I can see SS2K which I selected. Looks like it connected via Bluetooth as the light stopped flashing when I started qz.

So seems like first issue is figuring what I am doing wrong to get into the web page....

doudar, 2024-02-20:
To add to @marklogan17 , unless the SS2K is connected to your local WiFi, QZ probably needs to be off because of radio time switching limitations.

wassa22, 2024-02-20:
[image]

wassa22, 2024-02-20:
Yes that was closed...so reset again and have no issue getting to web page to set to my network... upon reboot I cannot get back into it....

doudar, 2024-02-20:
Are you connecting it to a 2.4ghz network?

doudar, 2024-02-20:
I'm very close to having a config app ready for this so that Wifi setup isn't needed. I'll probably ask Roberto if he can add the "none" and none" to his initialization routine for SmartSpin2k to automate this part of the setup once the associated firmware is ready.

wassa22, 2024-02-20:
Ok here is where I am at.... reset wirless access point and ensured only 2 4g was active.
Reset SS2K again and connected to network as well as changing name to see if qz picked it up which it did.
Have been in and out of webpage a few times on my phone no issue so far.
It looks like when I start qz it is connecting to SS2K via bluetooth as the blue light stops blonking

marklogan17, 2024-02-20:
Yes QZ will connect via Bluetooth. It's outside the interfaces configured in the ss2k Bluetooth page. Now you should try the physical buttons and the tiles as I described above

wassa22, 2024-02-20:
Have run through the list and steps as above.....the physical buttons on the resistance tile do not work. The gear one does but pushing that has no effect on the SSK2

wassa22, 2024-02-20:
@doudar so have had to change tack and connect mywhoosh to SS2k and use qz to provide power info. I cant seem to choose qz as power source but it is receiving the data. Only issue is the power in mywhoosh keeps dropping down to zero quite frequently, along with everynow and then my ipad pops up saying ss2k wants to pair. Unfortunately I cant use qz for power in mywhoosh as ut has to be the same as the controller.

On the plus side the SS2K is changing the resistance fine so I am def further along than I was this morning.

Ps...did have to reset ss2k again to get back to the web page this afternoon. 😁

wassa22, 2024-02-21:
@doudar will close this thread but leave the one open in qz Github as issue of getting to web page is resolved even though I still find it temperamental. Issue left to figure out is the power readings being sent...cheers.

closed 2024-02-21 by wassa22