doudar/SmartSpin2k #538: [BUG] Bluetooth Connection

Bofunga79, 2024-03-17:
[image]

### Description

I can't connect my Bluetooth power meter and heart rate strap with the latest firmware anymore. With each rescan, there are increasingly more entries. Although I can select them, none deliver values to Zwift.

I attempted a downgrade, which restored Bluetooth functionality, but afterward, I could no longer connect to the SSID (received a "wrong password" message). Consequently, I had to use the flash tool to reinstall the current firmware.

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* If you provide debug output and it is long, consider using [Pastebin](https://pastebin.com/) and paste a link in this issue to keep the issue easier to read. 
* Provide pictures or links to videos that clearly demonstrate the issue.

doudar, 2024-03-17:
Are you able to provide a log that shows the connection process?

I'm expecting a connection issue if you don't have the new device names selected (the ones with two extra digits than before the update), however it looks like you may have them selected. A log file would be great. 

I believe the issue could possibly also be because of using the web frontend with the new update. Possibly The web isn't handling the space correctly and truncating the name. A log would be great. 

You can also check and see if using the App fixes your issue. Please try testing it (after you get a log of your current setup). If you use Android, you can find the install .apk in the GitHub link below. If you use iOS, you can join the public beta here: https://testflight.apple.com/join/XvIbJXta
https://github.com/doudar/SS2kConfigApp/releases

doudar, 2024-03-17:
I just check my setup and the names (including the space) appear to be getting passed correctly (on Chrome) 

Looking forward to your log.

Bofunga79, 2024-03-17:
[log.txt](https://github.com/doudar/SmartSpin2k/files/14628490/log.txt)

I'm having the same problems with the app. Thanks in advance for your help :-)

doudar, 2024-03-17:
Great - Thanks for the log. It's not the problem I thought you were having. It's connecting just fine, however we never get sent any data. What's the last known working binary that works? Also, what bike and hrm is it?

Bofunga79, 2024-03-17:
When I downgrade to 24.2.27, the connection works flawlessly, but unfortunately, I can no longer connect with the SSID and the default "password". It is a Christopeit Racer Bike 4.0, identical in construction to the Yesoul S3 (probably just rebranded).

doudar, 2024-03-17:
Thanks - that will give me a place to start.

Is that a pre-config app version? Can you send me the direct link to the location of the binary you are using?

Bofunga79, 2024-03-17:
https://github.com/doudar/SmartSpin2k/releases/download/24.3.16/SmartSpin2kFirmware-24.3.16.bin.zip

https://github.com/doudar/SmartSpin2k/releases/download/24.2.27/SmartSpin2kFirmware-24.2.27.bin.zip

I just saw the new version 24.3.17, I will test it right away...

doudar, 2024-03-18:
I haven't found "the smoking gun".  See if this binary helps you at all. If not, it would be very helpful if you can connect a microusb and then use the smartspin2kflasher (or another serial terminal) to capture the enhanced logging I enabled in this build. This binary is also sending a FTMS "start training" command to the bike after connection is made. It's possible that may help.  

Make sure you unzip it before flashing :)
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/14643498/firmware.zip)

Bofunga79, 2024-03-19:
I've tested the firmware, but unfortunately, the problem remains the same. I'm attaching the log file.
[log2.txt](https://github.com/doudar/SmartSpin2k/files/14647069/log2.txt)

Bofunga79, 2024-03-19:
And additionally, here's a log from the firmware version 24.2.27, which works for me.
[log 24.2.27.txt](https://github.com/doudar/SmartSpin2k/files/14647211/log.24.2.27.txt)

doudar, 2024-03-19:
Okay - another small change. I increased the log level again and the possible fix is I lowered the MTU, which was increased for BLE firmware updates. If this works (I have hope) I think I might be able to make that setting firmware update specific. 
[firmware_noMTU.zip](https://github.com/doudar/SmartSpin2k/files/14651481/firmware_noMTU.zip)

Bofunga79, 2024-03-19:
I tested the firmware, unfortunately no improvement. I'm attaching the log again.
[log3.txt](https://github.com/doudar/SmartSpin2k/files/14654724/log3.txt)

doudar, 2024-03-19:
This is baffling me. 

It appears that the notifications are not getting enabled (or set) correctly on your board. 

Perhaps @H2zero would be kind enough to take a look at this latest log. ^^

The logs on my system (same firmware) all contain callbacks such as: 
`Descriptor: uuid: 0x2902, handle: 41
D NimBLEClient: Got Client event 
D NimBLEClient: Notify Recieved for handle: 32
D NimBLEClient: checking service 0x1826 for handle: 32
D NimBLEClient: Got Notification for characteristic Characteristic: uuid: 0x2ad2, handle: 32 0x0020, props:  0x12`

On your system, all I see in this place is:
`[19:13:48]D NimBLECharacteristicCallbacks: onNotify: default
[19:13:48]D NimBLEServer: >> handleGapEvent: 
[19:13:48]D NimBLECharacteristicCallbacks: onStatus: default`

In the meantime, just for kicks, I'll try rolling back the Arduino core and post a build here to see if that fixes it. 

Callbacks should be getting registered here: 
https://github.com/doudar/SmartSpin2k/blob/571febdcf9ddf5e39e97f1bdbd4948e3ceee51f6/src/BLE_Client.cpp#L287

doudar, 2024-03-19:
Okay - the onNotify: default lines seem to be coning from the server, not the client, so I think those aren't related.

doudar, 2024-03-19:
[firmware_with_Initial_read.zip](https://github.com/doudar/SmartSpin2k/files/14657685/firmware_with_Initial_read.zip)
Okay - give this a try. No need to connect to Zwift - it would be best if you only connect the device you're having problems with. 

Thanks! We'll eventually get this.

Bofunga79, 2024-03-20:
Wow, thanks again for the tireless effort. I tested and connected only the power meter, but unfortunately still no values...
[log4.txt](https://github.com/doudar/SmartSpin2k/files/14662875/log4.txt)

Bofunga79, 2024-03-20:
With the new firmware, I'm once again unable to log into the SSID using "password". However, pressing both shifter buttons successfully connects the power meter, and the values are displayed in Zwift. Really great :-)
[log5.txt](https://github.com/doudar/SmartSpin2k/files/14668090/log5.txt)

Bofunga79, 2024-03-20:
I configured the Wi-Fi through the app for a login to my SSID "Extern," but it seems he's not using it. Instead, it defaults back to the local AP mode and the SSID "Smartspin2k." I then tried changing just the password through the app, but that didn't work either. However, I can connect the power meter through the app, and it works fine. Personally, controlling it through the app would be enough for me...
[log6.txt](https://github.com/doudar/SmartSpin2k/files/14668669/log6.txt)

doudar, 2024-03-21:
#540 

@Bofunga79 , here is a current binary cleaned up and ready to be merged. I'd appreciate it if you can test it one more time. WiFi Should also be fixed. If you use the app to enter in your station password, I found a bug in the app (https://github.com/doudar/SS2kConfigApp/issues/29) - you'll need to press the enter key on the keyboard before you press save, otherwise it sends the previous value. If it's working, I don't need a log. 

Here is a binary
[firmware_Ready_To_Merge_1.zip](https://github.com/doudar/SmartSpin2k/files/14696565/firmware_Ready_To_Merge_1.zip)

Thank you!

Bofunga79, 2024-03-21:
Everything is now working as it should! Thanks again for your tireless effort :-)

doudar, 2024-03-23:
fixed by #540

closed 2024-03-23 by doudar