doudar/SmartSpin2k #545: [BUG] CSC sensor cadence incorrect

jekkos, 2024-04-18:
### Description

I have deployed the latest build that has CSC support. It seems that it's not working as it should. In zwift I see the value stuck at 70 or some value above it. The reading does work properly when I connect the bike directly to Zwift.

### Steps to reproduce

**Expected behavior**
To see a correct cadence reading in Zwift

**Actual behavior**
Value is stuck at 70-73 most of the time

### Additional Information

[smartspin2.log](https://github.com/doudar/SmartSpin2k/files/15030238/smartspin2.log)

jekkos, 2024-08-07:
[smartspin2.log](https://github.com/user-attachments/files/16534753/smartspin2.log)

@eMadman I wrote you about this issue some time ago on facebook. Today I have updated to the latest firmware, but still the CSC is not giving me any data back. Attached here a second logfile.
Any idea what the issue could be  @doudar ?

doudar, 2024-08-07:
If you go to releases, this should be fixed on the latest build. I'll be making that an automatic update in the next 24 hours.

doudar, 2024-08-07:
Also, if it doesn't work, please let me know.

closed 2024-11-11 by doudar

jekkos, 2024-11-11:
Will try again this week with the latest build but last time it did not work yet.

jekkos, 2024-12-26:
Have tested this again, but it still doesn't work. I'll try to add a new logfile here this week. I have flashed the very last version from github. Cadence reading sticks to zero while there is power output.

doudar, 2024-12-26:
Im guessing that this is because sensor collector is re-writing the cadence value to zero from either the connected power meter or from the connected fitness machine service.

jekkos, 2024-12-26:
So there might still be a bug? If I connect the sensor directly to Zwift I do see power and cadence values.

doudar, 2024-12-26:
I think I'm seeing your issue. CSC INPUT isn't supported yet. Only Output. For some reason your bike doesn't send the cycling cadence out through the Cycling power service. I'll see if I can implement CSC input today. It's been on my list.

jekkos, 2024-12-26:
Ok that would be really great!

jekkos, 2024-12-26:
I found this branch: https://github.com/doudar/SmartSpin2k/commits/Adding-CSC if there is something I can try, let me know.

jekkos, 2025-05-10:
Hi, is there any branch I can test? changes done here might not have made it to the master yet?

doudar, 2025-05-11:
the current testing and unit_testing branches will connect to CSC devices. Neither are completely stable/functional, but feel free to give them a try!

jekkos, 2025-05-12:
Anything I can help out with here?

doudar, 2025-05-12:
> Anything I can help out with here?

I've just sent some changes to Unit_Testing. I haven't done anything other than bench testing these changes, but you can try a couple things. Compile and upload Unit_Testing. Try to load a power table (manage power table menu), home your SmartSpin2k (using calibrate in an app), Connect your cadence sensor, and then enable use "power table for power". You should then get power output using a combination of knob position and cadence. It's not perfect (YET!), but I'm getting there. 

If you don't have a power table, here is one to try (download, unzip test.ptab, and then use "import" power table, then "load power table".

[Test.zip](https://github.com/user-attachments/files/20175905/Test.zip)

jekkos, 2025-05-30:
I have made a build of the latest code in UNit_Testing branch. So indeed I was able to connect my cadence sensor now which is great! I'll try to generate a power table first with my Assioma and do a couple of runs with it, to test if that base scenario is all stable. Once I have that then I'll try the cadence to power. 

I was looking in the web interface but have the impression the focus has shifted a bit towards the app, which I have also installed now.