doudar/SmartSpin2k #547: Erratic ERG in Trainer Day

aopenshaw, 2024-04-22:
-->
### Description

Begin workout in TrainerDay in ERG mode. After a short time ERG becomes disabled in TD screen and slope mode is the only option. During the remainder of the workout I use the shifter to control the resistance.  Sometimes during this interval, the shifter will let me shift up but not down so I have to use the manual knob to lower the resistance. I am also experiencing dropouts during the workout. I cannot return to ERG mode unless I disconnect and reconnect the bike. Sometimes I have to exit the app also.

### Steps to reproduce
Sometimes
1. Begin workout in TrainerDay
2. Use ERG mode until it becomes disabled.
3. [Continue with Slope mode for the duration

**Expected behavior**

Should be able to use ERG or slope mode during workout and switch back and forth

**Actual behavior**

ERG controls resistance for a while then becomes disabled

### Additional Information
I have contacted TrainerDay and they are willing to work with you to correct the problem. They are looking into it but are unfamiliar with SS2K. They wanted me to submit a ticket to you as well. They have not seen this behaviour on other trainers. Logs and image attached.

Schwinn IC4 
Tempo Power Meter on Left Crank connected through SS2K
Samsung Galaxy 7 FE Tablet
[sweet spot A.txt](https://github.com/doudar/SmartSpin2k/files/15065820/sweet.spot.A.txt)
[image]

aopenshaw, 2024-04-22:
> Looking at the logs, your erg targets are coming below 50 watts - which
> is the default minimum allowed by the SmartSpin2k.  This is a safety
> feature intended to prevent the knob from crashing to its physical
> limit.  You can lower this setting in the SS2K companion app or web
> config of the smartspin2k.
>
> Take the SmartSpin2k off your bike and lower your resistance to
> something approx 1-2 full turns from the physical lower limit of the
> knob.  Pedal at 90rpm and record the watts reported by your power
> meter.  Set this as your min brake watts.  That should get you going.

I will do that. Why does ERG not work even at higher targets? Thanks!

⁣Andy Openshaw

Sent from BlueMail ​

aopenshaw, 2024-04-22:
I just did another couple of short test rides at higher target wattage and the ERG still quits working sometime early in the ride. Minimum wattage is set at 55. Is there a particular test I could do to help diagnose? Thanks!

aopenshaw, 2024-04-22:
Did another short test with short target intervals going from 65w to 90w. I set the minimum wattage to 35 before I did this. Had one dropout at first but it reconnected and ERG stayed on for the full ride although it was a short one. Logs are attached. Thank you!
[04222024 logs 65w-90w.txt](https://github.com/doudar/SmartSpin2k/files/15068463/04222024.logs.65w-90w.txt)

aopenshaw, 2024-04-22:
Did another with the same intervals. I selected ERG and the indicator on the app said I was in ERG but it never took control. Logs attached.
[65w-90w-no control.txt](https://github.com/doudar/SmartSpin2k/files/15069135/65w-90w-no.control.txt)

aopenshaw, 2024-04-27:
Have you had a chance to test this? I am having really erratic behavior with Trainer Day. It seems to have problems controlling the device. I have never had any problems like it with any other app.

doudar, 2024-04-28:
Sorry - busy week. I just went through the code and am about to test it. I think I found a bug that might have been causing this: 
https://github.com/doudar/SmartSpin2k/blob/b2d7a88b34aaa750635b42ba626edc22b2ef6537/src/ERG_Mode.cpp#L575

Changed to:
https://github.com/doudar/SmartSpin2k/blob/e66162a93aeaa308883cfd3dc1d18800e8759063/src/ERG_Mode.cpp#L575

Attached is a firmware.bin if you would like to try it. Unzip first, then upload (preferably via wifi). User and pass are admin. firmware must be named exactly firmware.bin

[firmware.zip](https://github.com/doudar/SmartSpin2k/files/15140405/firmware.zip)

doudar, 2024-04-28:
I just test rode it and it's good on my end. I'm going to merge this change into develop

Fixed by #546

aopenshaw, 2024-04-28:
Thanks! I will give it a try tomorrow.

aopenshaw, 2024-04-28:
Tried a couple of tests and it seems to be tracking now. The only weird thing I noticed is that when I started a couple of workouts the resistance shot up to 50 at the beginning even though the target was 55 w. After pedaling through it it tracked back down though. Thank you and I will let you know after a few rides how its going but I hope this has fixed the problem in TD.

eMadman, 2024-04-28:
> Tried a couple of tests and it seems to be tracking now. The only weird thing I noticed is that when I started a couple of workouts the resistance shot up to 50 at the beginning even though the [...]

What was your cadence that triggered the high resistance?  Generally speaking, this is behavior you'll encounter at a too low cadence because the smartspin2k is trying to bring you up to the target wattage.

aopenshaw, 2024-04-28:
I usually start the workout and then start pedaling so zero at first and quickly ramped up to 85 but by that time it shot up. So are saying there is no delay at the beginning of a workout? Should I have the cadence up before I begin the workout?

aopenshaw, 2024-04-29:
Had lots of problems in TrainerDay this morning. Unfortunately I did not get all of the logs but I did pause a couple of times to start some logs. It never would track in ERG mode. I started the workout and the first interval I came to it remained at the previous resistance.  I paused and started the logging then.

 In the first log it just kept chasing from resistance 7 to 60 up and down and never settled down. I disconnected and reconnected again and set it in ERG mode and it never tracked at all...just acted like it was in incline mode.

[Custom UDP Receiver (Port 10000)0429-2.txt](https://github.com/doudar/SmartSpin2k/files/15150807/Custom.UDP.Receiver.Port.10000.0429-2.txt)
[Custom UDP Receiver (Port 10000)0429-1.txt](https://github.com/doudar/SmartSpin2k/files/15150808/Custom.UDP.Receiver.Port.10000.0429-1.txt)

aopenshaw, 2024-04-30:
Hey unfortunately I'm still having major problems. When you have a chance can you take a look at my logs and see what's going on? 

Thanks!

⁣Andy Openshaw

Sent from BlueMail ​

eMadman, 2024-04-30:
> Hey unfortunately I'm still having major problems. When you have a chance can you take a look at my logs and see what's going on? Thanks! ⁣Andy Openshaw Sent from BlueMail ​
> […](#)
1533121)

My test ride last night failed in spectacular fashion.  I'm running into issues with the Trainer Day app not sending data to my SmartSpin2k.  My early theory is that the Trainer Day app loses connection to the device if the app loses focus but it's hard to say right now.  In-app, the smartspin2k looks connected but nothing's being sent.  

This will need more time to test and reproduce.  It may not be an overnight fix.

In the meantime, can you export the trainer day workouts to be used in another app?  [Qdomyos-Zwift](https://www.qzfitness.com) is a good option for riding the exported .zwo workouts.  [Flux ](https://github.com/dvmarinoff/Flux) is an interesting option as well but it's very dependent on your web browser.

doudar, 2024-05-01:
Just checking in - I have a very busy couple of weeks here. I’m paying attention, but it will be a bit before I can look at this more.

doudar, 2024-05-01:
Yah, browsing the forums, I see a lot of little issues there and I think they’re all seen in one way or another. 

On my testing that worked, I force closed the app, started a workout, kept the app in the foreground the entire time. This was on iOS. 

Then I had some other testing to do and so I closed the app, opened Zwift … and TrainerDay never released control, so I needed to force quit TrainerDay again. 

So, from my testing, I think it works, but I think their app has a lot of issues when it loses focus. Which, on iOS, I understand because iOS multitasking isn’t great.

closed 2024-07-29 by doudar