doudar/SmartSpin2k #550: [BUG] Disconnects with  Tempo PM

eMadman, 2024-05-18:
User reporting disconnects with tempo power meter.  

This log file comes off the timing-updates branch with timeouts set to pClient->setConnectionParams(6, 12, 0, 495);
[Custom UDP Receiver (Port 10000)051024-1 TD disconnects.txt](https://github.com/doudar/SmartSpin2k/files/15364561/Custom.UDP.Receiver.Port.10000.051024-1.TD.disconnects.txt)
```
[1919695][E](BLE_Server): 64 02 eb 05 84 00 00 00 16 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(66.00) PW(22) SD(15.00) ]
[1919696][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1920209][E](BLE_Server): 64 02 eb 05 84 00 00 00 16 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(66.00) PW(22) SD(15.00) ]
[1920209][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1920734][E](BLE_Common): 2e 0f 00 00 7f da b8 09 73 b0 <- 0x1818 | 0x2a63 | CPS[ CD(66.57) PW(0) POS(-1100) ]
[1920735][E](BLE_Common): 00 68 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(104) POS(-1100) ]
[1920741][E](BLE_Server): 64 02 00 00 84 00 00 00 00 00 68 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(104) CD(66.00) PW(0) SD(0.00) ]
[1920741][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1921249][E](BLE_Common): 00 68 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(104) POS(-1100) ]
[1921251][E](BLE_Server): 64 02 00 00 84 00 00 00 00 00 68 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(104) CD(66.00) PW(0) SD(0.00) ]
[1921252][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1921763][E](BLE_Server): 64 02 00 00 84 00 00 00 00 00 68 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(104) CD(66.00) PW(0) SD(0.00) ]
[1921763][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1922271][E](BLE_Common): 00 68 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(104) POS(-1100) ]
[1922291][E](BLE_Server): 64 02 00 00 84 00 00 00 00 00 68 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(104) CD(66.00) PW(0) SD(0.00) ]
[1922292][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1922805][E](BLE_Common): 2e 0f 00 00 7f da b9 09 62 b8 <- 0x1818 | 0x2a63 | CPS[ CD(30.25) PW(0) POS(-1100) ]
[1922806][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 bc <- 0x1818 | 0x2a63 | CPS[ CD(30.25) PW(0) POS(-1100) ]
[1922812][E](BLE_Server): 64 02 00 00 3c 00 00 00 00 00 68 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(104) CD(30.00) PW(0) SD(0.00) ]
[1922813][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1923320][E](BLE_Common): 00 68 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(104) POS(-1100) ]
[1923322][E](BLE_Server): 64 02 00 00 3c 00 00 00 00 00 68 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(104) CD(30.00) PW(0) SD(0.00) ]
[1923323][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1923832][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 c0 <- 0x1818 | 0x2a63 | CPS[ CD(30.25) PW(0) POS(-1100) ]
[1923835][E](BLE_Server): 64 02 00 00 3c 00 00 00 00 00 68 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(104) CD(30.00) PW(0) SD(0.00) ]
[1923836][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1924344][E](BLE_Common): 00 69 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(105) POS(-1100) ]
[1924348][E](BLE_Server): 64 02 00 00 3c 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(30.00) PW(0) SD(0.00) ]
[1924350][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1924870][E](BLE_Server): 64 02 00 00 3c 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(30.00) PW(0) SD(0.00) ]
[1924871][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1925382][E](BLE_Common): 00 69 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(105) POS(-1100) ]
[1925384][E](BLE_Server): 64 02 00 00 3c 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(30.00) PW(0) SD(0.00) ]
[1925385][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1925894][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 c4 <- 0x1818 | 0x2a63 | CPS[ CD(30.25) PW(0) POS(-1100) ]
[1925895][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 c8 <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) POS(-1100) ]
[1925901][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1925902][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1926411][E](BLE_Common): 00 69 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(105) POS(-1100) ]
[1926421][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1926422][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1926931][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 cc <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) POS(-1100) ]
[1926934][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1926935][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1927450][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1927451][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1927963][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1927964][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1928473][E](BLE_Common): 00 69 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(105) POS(-1100) ]
[1928481][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1928483][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1929001][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 d0 <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) POS(-1100) ]
[1929002][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 d4 <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) POS(-1100) ]
[1929008][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1929009][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1929521][E](BLE_Common): 2e 0f 00 00 7f da b9 09 00 d8 <- 0x1818 | 0x2a63 | CPS[ CD(0.00) PW(0) POS(-1100) ]
[1929521][E](BLE_Common): 00 69 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(105) POS(-1100) ]
[1929526][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 69 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(105) CD(0.00) PW(0) SD(0.00) ]
[1929527][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1930035][E](BLE_Common): 00 67 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(103) POS(-1100) ]
[1930038][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 67 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(103) CD(0.00) PW(0) SD(0.00) ]
[1930039][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1930557][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 67 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(103) CD(0.00) PW(0) SD(0.00) ]
[1930558][E](FTMS_SERVER): 11 00 00 2c 01 28 00 -> Sim Mode Incline 3.000000
[1931066][E](BLE_Common): 00 67 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(103) POS(-1100) ]
[... 162 lines cut ...]
[695515][E](BLE_Server): 30 00 00 00 2d 03 0f 93 2d 03 00 00 5c e3 -> 0x1818 | 0x2a63 | CPS(CPM)[ CD(0.00) PW(0) ]
[695522][E](BLE_Server): 00 65 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(101) ]
[695528][E](FTMS_SERVER): 05 19 00 -> ERG Mode Target: 25 Current: 0 Incline: -14.380000
[695564][E](ERG_Mode): TorqueTable pairs [1][2]
[695565][E](ERG_Mode): Min Position Set: -2255
[695566][E](ERG_Mode): TorqueTable pairs [2][1]
[695567][E](ERG_Mode): Max Position Set: 11405
[696037][E](BLE_Server): 64 02 00 00 00 00 00 00 00 00 65 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(101) CD(0.00) PW(0) SD(0.00) ]
[696040][E](BLE_Server): 30 00 00 00 2d 03 0f 93 2d 03 00 00 5c e3 -> 0x1818 | 0x2a63 | CPS(CPM)[ CD(0.00) PW(0) ]
[696043][E](BLE_Server): 00 65 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(101) ]
[696044][E](BLE_Server): 03 2d 03 00 00 5c e3 2d 03 0f 93 -> 0x1816 | 0x2a5b | CSC(CSM)[ WheelRev(813) WheelTime(38797312) CrankRev(1091800435) CrankTime(813) ]
[696550][E](BLE_Common): 2e 5f 31 00 e9 0c 88 02 58 56 <- 0x1818 | 0x2a63 | CPS[ CD(88.50) PW(49) POS(-1438) ]
[696551][E](BLE_Common): 00 65 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(101) POS(-1438) ]
[696558][E](BLE_Server): 30 00 31 00 2e 03 c9 95 2e 03 00 00 e8 e4 -> 0x1818 | 0x2a63 | CPS(CPM)[ CD(88.00) PW(49) ]
[696560][E](BLE_Server): 00 65 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(101) ]
[696967][E](TTable): Target Slot (6n.M)(1)(-1446) was greater than next (2)(-1764)
[696968][E](TTable): |    0|    7|    0|   14|   21|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|    0|
[... cut ...]    0|    0|    0|    0|    0|    0|    0|    0|
[697023][E](ERG_Mode): TorqueTable pairs [1][-1]
[697068][E](BLE_Common): 00 65 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(101) POS(-1438) ]
[697069][E](BLE_Server): 64 02 bb 07 b0 00 00 00 31 00 65 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(101) CD(88.00) PW(49) SD(19.00) ]
[697071][E](BLE_Server): 30 00 31 00 2f 03 84 98 2f 03 00 00 75 e6 -> 0x1818 | 0x2a63 | CPS(CPM)[ CD(88.00) PW(49) ]
[697074][E](BLE_Server): 00 65 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(101) ]

```

doudar, 2024-11-11:
Have we had any recent updates or activity on this?

doudar, 2025-01-24:
Closing due to inactivity

closed 2025-01-24 by doudar