doudar/SmartSpin2k #559: [BUG] Unable to mount LittleFS after flashing firmware on new PCB, mounts fine after re-flashing with mobile app instead.

jrbarrick, 2024-07-28:
### Description

Unable to mount LittleFS after flashing firmware on new PCB, mounts fine after re-flashing with mobile app.  Reproduced on 3 new PCBs.  Once the issue is fixed by flashing with the mobile app, flashing later on over USB works fine.

### Steps to reproduce

1. Flashed latest firmware (24.7.9) to ESP32 over USB using the SmartSpin2kFlasher tool.  Flash reports that it works fine and hashes verify. Device reboots fine, however errors are thrown related to LittleFS:
```
Using 'COM12' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..........
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: A0:A3:B3:79:88:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001adfff...
Compressed 15872 bytes to 10319...
Writing at 0x00001000... (100 %)
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.6 seconds (effective 230.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 313.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 442.3 kbit/s)...
Hash of data verified.
Compressed 1693792 bytes to 1034329...
Writing at 0x00010000... (1 %)
...
Writing at 0x001acb9e... (100 %)
Wrote 1693792 bytes (1034329 compressed) at 0x00010000 in 26.6 seconds (effective 509.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[15:14:10][    30][E](Main): Compiled Apr 28 202401:49:10
[15:14:10][    31][E](Main): Current Board Revision is: Revision Two
[15:14:10][    33][E](Main): Mounting Filesystem
[15:14:10]./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
[15:14:10]E (28) esp_littlefs: mount failed,  (-84)
[15:14:10]E (28) esp_littlefs: Failed to initialize LittleFS
[15:14:10][    36][E][LittleFS.cpp:95] begin(): Mounting LittleFS failed! Error: -1
[15:14:10][    38][E](Main): An Error has occurred while mounting LittleFS.
[15:14:10][    41][E](FS_Upgrader): Stopping LittleFS
[15:14:10]E (34) SPIFFS: mount failed, -10025
[15:14:10][    42][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
[15:14:10][    42][E](FS_Upgrader): SPIFFS Not Found - Could not upgrade FS
[15:14:10]./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
[15:14:10]E (139) esp_littlefs: mount failed,  (-84)
[15:14:10]E (139) esp_littlefs: Failed to initialize LittleFS
[15:14:11][   653][E](Config): Reading File: /config.txt
[15:14:11][   655][E][vfs_api.cpp:105] open(): /littlefs/config.txt does not exist, no permits for creation
[15:14:11][   655][E](Config): Couldn't find configuration file.
[15:14:11][   658][E](Config): Contents of file: /config.txt
[15:14:11][   659][E][vfs_api.cpp:105] open(): /littlefs/config.txt does not exist, no permits for creation
[15:14:11][   659][E](Config): Failed to read file
[15:14:11][   662][E][vfs_api.cpp:182] remove(): /config.txt does not exists or is directory
[15:14:11][   663][E](Config): Writing File: /config.txt
[15:14:11][   678][E](Config): Reading File: /userPWC.txt
[15:14:11][   681][E][vfs_api.cpp:105] open(): /littlefs/userPWC.txt does not exist, no permits for creation
[15:14:11][   682][E](Config): Couldn't find configuration file. Loading Defaults
[15:14:11][   684][E](Config): Contents of file: /userPWC.txt
[15:14:11][   687][E][vfs_api.cpp:105] open(): /littlefs/userPWC.txt does not exist, no permits for creation
[15:14:11][   687][E](Config): Failed to read file
[15:14:11][   692][E][vfs_api.cpp:182] remove(): /userPWC.txt does not exists or is directory
[15:14:11][   692][E](Config): Writing File: /userPWC.txt
```
3. Can connect just fine with BT, and app connects to device fine.  Settings can be "saved" in the app, however they aren't saved and are lost at reboot.
4. Can retry flashing firmware over USB with no change, and also tried 24.4.28 over USB.
5. However, flashing 24.4.28 using the Firmware Update option in the mobile app seems to work different, as partition mounts fine when rebooting after flash using the app, and settings can be saved without any issues:
```
[15:17:39]1. BeginOTA2. Running partition type 0 subtype 16 (offset 0x00010000) 
[15:17:39]3. Writing to partition subtype 17 at offset 0x1f0000 
...
[15:20:20]..................................................................................4. Final byte arrivedSet Boot partionRestarting...
[15:20:20][    31][E](Main): Current Board Revision is: Revision Two
[15:20:20][    33][E](Main): Mounting Filesystem
[15:20:20][    38][E](Config): Reading File: /config.txt
[15:20:20][    60][E](Config): Config File Loaded: /config.txt
[15:20:20][    61][E](Config): Contents of file: /config.txt
[15:20:20][    86][E](Config): Writing File: /config.txt
[15:20:20][   110][E](Config): Reading File: /userPWC.txt
[15:20:20][   135][E](Config): Config File Loaded: /userPWC.txt
[15:20:20][   135][E](Config): Contents of file: /userPWC.txt
[15:20:20][   171][E](Config): Writing File: /userPWC.txt
```

**Expected behavior**
Firmware boots and mounts LittleFS

**Actual behavior**
Firmware boots and fails to mount LittleFS

### Additional Information
Full log:
https://pastebin.com/eMT0VqZS

doudar, 2024-07-28:
This might be more of an issue with documentation. 

On initial installs, in PlatformIO, you need to select "upload file system image" as well. 

The upgrade probably works because there's a littlefs.format() in order to fix errors like this.

In new branches I removed the spiffs upgrader (firmwares that were over two years old used spiffs). Instead of the upgrader, I suppose I could place a format()  on boot if the files cannot be accessed. 

Thanks for the report!

jrbarrick, 2024-07-28:
Read the LittleFS docs a bit as well, looks like you can just call LittleFS.begin(true) instead of LittleFS.begin(false) and it will format the LittleFS partition if it doesn't exist.  Did some testing on my end and it seems like that does the trick.

doudar, 2024-07-28:
I'm guessing it was set to (false) previously for the upgrade logic to work correctly. I think for now I slightly prefer keeping the format() inside of the if statement, just for readability. 

I'd love any more insight you might have! Thanks for the help!

commit 65a00455 (2024-07-28): fixes #559
closed 2024-07-28 by doudar