doudar/SmartSpin2k #561: [BUG] Sifter stopped working after Qdomyos

Dekto, 2024-08-04:
### Description

After using the Qdomyos Zwift app for automatic control during Peloton workouts on the Peloton Bike, the shifter no longer works.

### Steps to reproduce

1. Start Peloton Bike / Tablet
2. Start SS2k
3. Start Qdomyos Zwift App
4. Start a Peloton Workout on the Bikes Tablet
5. Confirm Popup on Qdomyos Zwift App
6. Complete workout
7. Close Qdomyos Zwift App

**Expected behavior**

The resistance should change via the shifter.

**Actual behavior**

When you tap the shifter button, the SS2k LED lights up but doesn't respond.

### Additional Information

Peloton Bike
iPhone SE (2022) iOS 17.5.1
SS2k 24.4.28
Qdomyos Zwift 2.15.4

eMadman, 2024-08-04:
I'll try to reproduce this on my bike this weekend.

doudar, 2024-08-09:
@cagnulein Can you send 0x01,0x1C (reboot) to the SS2k at the end of a session? That should resolve this issue. I'm guessing it's because of external control being used.  

https://github.com/doudar/SmartSpin2k/blob/69961f3d5a8e6e8601faff27c9571bcece35ce09/src/BLE_Custom_Characteristic.cpp#L535-L542

cagnulein, 2024-08-09:
yes sure, let me try to create a branch for this @Dekto @doudar

doudar, 2024-08-09:
At the start of a session, everything should be the same. 

Maybe this is specific to Peloton workouts. What commands are you using for control on the Peloton workouts?

cagnulein, 2024-08-09:
in order to answer you i need to understand if @Dekto is using SS2K in QZ like an ftms bike or like a ss2k device. I guess the first one (it has more sense indeed).

If it's so, i'm just sending FTMS_SET_TARGET_RESISTANCE_LEVEL command from 0 to 100

doudar, 2024-08-09:
^^ Using the BLE FTMS Characteristic.

cagnulein, 2024-08-09:
done @Dekto 
send me an email to roberto.viola83 at gmail.com indicating this ticket and I will send you the new beta. Thanks

Dekto, 2024-08-09:
@cagnulein Hello, I use it as a SmartSpin2k device in the QZ app.

Dekto, 2024-08-10:
@cagnulein 

Hello, unfortunately the beta V2.16.67 does not fix the problem...

doudar, 2024-08-10:
@Dekto , thanks. @eMadman , can you get a log this weekend? Probably the most important is what it's saying when you try to shift.

cagnulein, 2024-08-10:
Thanks, let me know if I have to revert it

Roberto Viola
Software engineer and open source enthusiast
http://robertoviola.cloud

eMadman, 2024-08-10:
@cagnulein Entirely unrelated issue on my end - i'm having difficulty getting a connection to it at all.  It seems to keep hanging on another device.  I'm confident I have something wrong in my settings.  

[debug-Sat_Aug_10_13_18_37_2024.log](https://github.com/user-attachments/files/16570395/debug-Sat_Aug_10_13_18_37_2024.log)

Dekto, 2024-08-10:
I could also upload my SmartSpin2k logs if I get information about where it is stored?

cagnulein, 2024-08-10:
> @cagnulein Entirely unrelated issue on my end - i'm having difficulty getting a connection to it at all. It seems to keep hanging on another device. I'm confident I have something wrong in my [...]

here i can see that qz is correctly connected to your ss2k. It's sending only resistance 25 and the other metrics are all 0. Is it intended?

doudar, 2024-08-10:
> I could also upload my SmartSpin2k logs if I get information about where it is stored?

Follow these instructions, and if you can, use a pastebin.com url to post them here. 

Thanks!

Dekto, 2024-08-10:
Ok, that's more complex than I thought. I'll try to figure it out in the next few days...

eMadman, 2024-08-12:
I figured out the QZ issue - it was my settings.

I was able to reproduce the issue.  

QZ Log attached: [debug-Sun_Aug_11_21_46_30_2024.log](https://github.com/user-attachments/files/16576841/debug-Sun_Aug_11_21_46_30_2024.log)

SmartSpin2k log:

```
	121883	[597319][E](BLE_Server): 30 00 00 00 89 03 00 00 99 8d 88 03 73 b6 -> 0x1818 | 0x2a63 | CPS(CPM)[ CD(0.00) PW(0) ]
	121884	[597320][E](BLE_Server): 03 89 03 00 00 99 8d 88 03 73 b6 -> 0x1816 | 0x2a5b | CSC(CSM)[ WheelRev(905) WheelTime(1470889984) CrankRev(1092688691) CrankTime(904) ]
	121885	[597327][E](BLE_Server): 64 02 00 00 00 00 23 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]
	121886	[597747][E](BLE_Server): Client ID: 0 Address: 59:77:1c:fb:d8:fe Unsubscribed to 0x2a5b
	121887	[597748][E](BLE_Server): Client ID: 0 Address: 59:77:1c:fb:d8:fe Unsubscribed to 0x2a63
	121888	[597751][E](BLE_Server): Client ID: 0 Address: 59:77:1c:fb:d8:fe Unsubscribed to 0x2a37
	121889	[597751][E](BLE_Server): Client ID: 0 Address: 59:77:1c:fb:d8:fe Unsubscribed to 0x2ad9
	121890	[597754][E](BLE_Server): Client ID: 0 Address: 59:77:1c:fb:d8:fe Unsubscribed to 0x2ad2
	121891	[597754][E](BLE_Server): Bluetooth Remote Client Disconnected. Remaining Clients: 0
	121892	[598705][E](Custom_C): 01 17 <-shifterPosition
	121893	[599680][E](Custom_C): 01 17 <-shifterPosition
	121894	[600120][E](Custom_C): 01 17 <-shifterPosition
	121895	[600559][E](Custom_C): 01 17 <-shifterPosition
	121896	[601363][E](Custom_C): 01 17 <-shifterPosition
	121897	[601812][E](Custom_C): 01 17 <-shifterPosition
	121898	[602251][E](Custom_C): 01 17 <-shifterPosition
	121899	[603129][E](Custom_C): 01 17 <-shifterPosition
	121900	[603577][E](Custom_C): 01 17 <-shifterPosition
	121901	[604089][E](Custom_C): 01 17 <-shifterPosition
	121902	[605040][E](Custom_C): 01 17 <-shifterPosition
	121903	[605479][E](Custom_C): 01 17 <-shifterPosition
	121904	[605927][E](Custom_C): 01 17 <-shifterPosition
	121905	[606878][E](Custom_C): 01 17 <-shifterPosition
	121906	[607317][E](Custom_C): 01 17 <-shifterPosition
	121907	[607982][E](Custom_C): 01 17 <-shifterPosition
	121908	[608567][E](Custom_C): 01 17 <-shifterPosition
	121909	[609006][E](Custom_C): 01 17 <-shifterPosition
```

doudar, 2024-08-12:
Interesting, The only time that line should print is when it gets a read command from the custom characteristic. Could QZ be sending a read command a ton of times on disconnect?

https://github.com/doudar/SmartSpin2k/blob/69961f3d5a8e6e8601faff27c9571bcece35ce09/src/BLE_Custom_Characteristic.cpp#L464-L479 

In any case, a read command shouldn't cause this issue, but it might be a spot to start looking.

cagnulein, 2024-08-12:
qz, when acts as a ftms bike as @eMadman settings, it doesn't read anything. it just uses the notifications, so it sounds very strange. maybe SS2K accepts 2 connections on bluetooth and it's connected to something else?

eMadman, 2024-08-13:
> qz, when acts as a ftms bike as @eMadman settings, it doesn't read anything. it just uses the notifications, so it sounds very strange. maybe SS2K accepts 2 connections on bluetooth and it's connected to something else?

Hmm.  There should be no other connection to the smartspin.  

Last night, the only way I got it to connect to the SmartSpin2k was to have it as an FTMS bike.  DIsabling that setting today, it seems to get a connection.  I'll capture a new log

eMadman, 2024-08-13:
Steps taken this time.  Motor doesn't turn when expected

QZ Log: [debug-Mon_Aug_12_21_37_23_2024.log](https://github.com/user-attachments/files/16593487/debug-Mon_Aug_12_21_37_23_2024.log)
SS2K Log: [Custom UDP Receiver (Port 10000).txt](https://github.com/user-attachments/files/16593511/Custom.UDP.Receiver.Port.10000.txt)

- Start SmartSpin2k and QZ
- Start Peloton ride
- End ride
- Quit app
- Press physical shifters up/down (421797)
- Try web shifter (138341)
- Relaunch QZ - new log file: [debug-Mon_Aug_12_21_44_42_2024.log](https://github.com/user-attachments/files/16593489/debug-Mon_Aug_12_21_44_42_2024.log)
- Try to shift in Ss2k app (139919 I think) 

- Try to  change resistance in QZ

cagnulein, 2024-08-13:
@eMadman ftms bike setting disabled or not doesn't change anything since you have the peloton ss2k setting enabled. so i guess it was something else.

qz in the log here just writes these as expected 

`Mon. Aug. 12 21:39:15 2024 1723513155133 Debug: ../src/devices/bluetooth.cpp void bluetooth::debug(const QString &) " >> 00 // requestControl"
Mon. Aug. 12 21:39:15 2024 1723513155451 Debug: ../src/devices/bluetooth.cpp void bluetooth::debug(const QString &) " >> 07 // start simulation"
Mon. Aug. 12 21:39:15 2024 1723513155839 Debug: ../src/devices/bluetooth.cpp void bluetooth::debug(const QString &) " >> 04 19 // forceResistance 25"
Mon. Aug. 12 21:39:59 2024 1723513199607 Debug: ../src/devices/bluetooth.cpp void bluetooth::debug(const QString &) " >> 04 1b // forceResistance 27"
Mon. Aug. 12 21:40:30 2024 1723513230488 Debug: ../src/devices/bluetooth.cpp void bluetooth::debug(const QString &) " >> 04 1d // forceResistance 29"
Mon. Aug. 12 21:43:12 2024 1723513392334 Debug: ../src/devices/bluetooth.cpp void bluetooth::debug(const QString &) " >> 04 2d // forceResistance 45"`

@doudar did you get something from the ss2k log?

doudar, 2024-11-10:
I'd appreciate another test. #591 should fix this.