doudar/SmartSpin2k #573: [FR] Always send resistance over FTMS Indoor Bike

doudar, 2024-09-13:
**Is your feature request related to a problem? Please describe.**
Even when we are not connected to a resistance reporting bike, FTMS sets the has resistance flag, which causes issues when an app expects the resistance number to change.

**Describe the solution you'd like**
Calculate a resistance from 0-100 for bikes that don't output resistance.

**Describe alternatives you've considered**
Optionally set the has resistance flag to zero when a bike doesn't report resistance. 

**Additional context**
Thanks @cagnulein for helping to identify this issue.

doudar, 2024-11-11:
#585

cagnulein, 2024-12-10:
@doudar any news about that? i would like to use this feature with some your users if possible

closed 2025-09-22 by doudar