doudar/SmartSpin2k #594: Engine is disabled in certain occasions with version 24.11.10

serbasan, 2024-11-14:
I'm trying out the latest versions and I've noticed that on 22.11.10, before starting a race, during warm-up at the start line, the engine cuts out and I lose power (my setup has a spring that returns to gear 1 if the engine cuts out).
Also during today's race, halfway through the race it happened again on one occasion that the engine cut out and I had to quickly shift up a gear to regain power.
On other versions this doesn't happen.

To reproduce the problem, you would:
1)Start a race, group training, etc. and spend some time at the starting line warming up.
2) Go up to a higher gear.
3) Wait a while for the engine to unlock and become free.

The correct behavior would be that the motor would always be powered and not unlocked as happens in other versions.

I don't know if this is the expected behavior from now on or if it is a version issue.
If there is anything I am willing to try and help.
Regards.

doudar, 2024-11-14:
My guess is that it's because I decreased the hold multiplier on this line in order to keep the stepper running cooler. 

https://github.com/doudar/SmartSpin2k/blob/83ac674679e998acc494d78ca038e52d0d5dac90/src/Main.cpp#L557

Try removing the *.1 and see if it fixes your issue.

doudar, 2024-11-14:
Actually, this could also be your specific issue. I removed the test for a client to be connected, and now use movement only. I believe after ~two minutes of stepper movement it de-powers

doudar, 2024-11-14:
That would be :
https://github.com/doudar/SmartSpin2k/blob/83ac674679e998acc494d78ca038e52d0d5dac90/src/Main.cpp#L471-L472

You could add a test for if client connected, 
stepper.autoenable(false);
stepper.enableOutputs(true);
else{
stepper.autoenable(true);
}

serbasan, 2024-11-14:
Hi. I've made those changes to make it the same as in previous versions and it doesn't happen anymore. The gears are kept in place since the motor is not disabled. Will you leave or remove this feature?

For testing, I rolled back to version 24.7.9 and now in the newer versions 24.11.X the gear indicator no longer appears in Zwift.

I have reinstalled the ESP32 with cable but I still haven't been able to get Zwift_Gear_UI to appear. Any ideas on how to get it back?

Thanks.

doudar, 2024-11-15:
#596 Adds a check so that the stepper won't de-power if cadence is detected.

doudar, 2024-11-16:
@serbasan , it seems that Zwift either removed the gear indicator or it's a bug. I know they've broken it in previous updates, so hopefully it's just a bug. Search on the Zwift forums for wattbike gear indicator.

serbasan, 2024-11-17:
Tested today and the motor has not disconnected at any time. As for me, this thread can be closed.
Regarding the other issue I mentioned, with 1.78 it seems that Zwift has changed something that affects how you transmit gears. In the 1.78 news forum there is someone who complains but it is because they are using an ant+ antenna and SmartSpin2k support is not provided :(
We will keep an eye out if it appears again. It is a shame because it was very good in races to see the gear you are in.
Thank you very much for your work.

doudar, 2024-11-17:
Good news!

I'm guessing the gear indicator will come back again. I'm seeing lots of wattbike complaints and this has been an ongoing issue with Zwift.