doudar/SmartSpin2k #605: [BUG] Bowflex IC bike SE stops working with latest release.

liuwh08, 2024-12-12:
See discussion https://github.com/doudar/SmartSpin2k/discussions/598

### Steps to reproduce

Flash the latest firmware and hook smartspin2k to zwift

**Expected behavior**
SmartSpin2K report power and cadence.

**Actual behavior**
SmartSpin2K report zero on power and cadence.

### Additional Information
* https://github.com/doudar/SmartSpin2k/discussions/598 cantains the logs as well for firmwares.
* 24.2.16 works but with some quirks.
* I also found connecting to GT Bike V would crash the game with 24.2.16 not sure if this is related though.

doudar, 2024-12-15:
Does it reliably connect and then disconnect a short time later?

liuwh08, 2024-12-15:
From bike console, it's still connected. However from the logs of smartspin2k, it doesn't print the reported Power and Cadence like older version.

liuwh08, 2024-12-16:
@doudar Is there a doc to build the firmware so I could test this out?

(I could probably also bi-sec to see which version this regression started).

doudar, 2024-12-16:
It's pretty straightforward once you install the PlatformIO extension into VScode. It should mostly self configure once you do that. Otherwise, the official task list is here:
https://github.com/doudar/SmartSpin2k?tab=readme-ov-file#contributing

doudar, 2024-12-16:
If you'd like to do uploading via USB directly from within PlatformIO/VScode, you'll probably need the Silabs driver as well (try it first).
https://www.silabs.com/developer-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads

madxime, 2024-12-22:
I'm not particularly familiar with bluetooth, but I was trying to find out what's going. If I connect to my IC bike SE from my PC with an app like LightBlue, I can subscribe to the 0x2AD2 UUID from service 0x1826 UUID. It sends which me a notification every 500 ms. 

When I look at the logs from SmartSpin2k, the subscription to the same UUID seems to work, but onNotify() from src/BLE_Client.cpp never gets called. As a consequence, collectAndSet() from src/SensorCollector.cpp also never gets called, so everything's always zero.

Let me know if there's something I can do to try to help you find the root cause of this.

madxime, 2024-12-22:
I managed to make it work again by commenting out two lines below in `src/BLE_Client.cpp`.
```
          // If we would like to control an external FTMS trainer. With most spin bikes we would want this off, but it's useful if you want to use the SmartSpin2k as an
          // appliance.
          if (userConfig->getFTMSControlPointWrite()) {
            writeCharacteristic->writeValue(FitnessMachineControlPointProcedure::RequestControl, 1);
            vTaskDelay(BLE_NOTIFY_DELAY / portTICK_PERIOD_MS);
            SS2K_LOG(BLE_CLIENT_LOG_TAG, "Activated FTMS Training.");
          }
          // writeCharacteristic->writeValue(FitnessMachineControlPointProcedure::StartOrResume, 1);
          SS2K_LOG(BLE_CLIENT_LOG_TAG, "Updating Connection Params for: %s", _BLEd.peerAddress.toString().c_str());
          // BLEDevice::getServer()->updateConnParams(pClient->getConnId(), 120, 120, 2, 1000);
          spinBLEClient.handleBattInfo(pClient, true);
```

For some reason, writing `FitnessMachineControlPointProcedure::StartOrResume` stops the notifications from being sent. Would it be better to request control first, then send StartOrResume? There's also some timing issues involved here if `StartOrResume ` is sent immediately after `RequestControl`, so it seems like it needs to wait for the `RequestControl` indication first. I'm asking these questions just to try to figure out what would be the best solution in your opinion because the IC bike SE doesn't actually need to write anything in the Fitness Machine Control Point.

doudar, 2024-12-22:
Ah...Yes...I remember solving that at one point....The issue came up once I started passing through FTMS control point writes to a connected device. It's useful for some cases.  From wat I remember, if the IC SE gets a command it doesn't support, it stops responding. The best solution would be to look at the fields for the Characteristic, and then only if the FTMS device connected supports control (incline, resistance, simulation, etc) we would attempt to pass through the control commands. Optionally, we could do what I did before and completely disable write support (or add a button on the app to enable write/passthrough support). The latter would be the easy solution, but I'm really trying to keep the number of settings to a minimum. I'd like SS2k to be as "plug and play" as possible. Nice work so far! I really appreciate your interest.

madxime, 2024-12-22:
You're welcome! I hope I understood properly how the control point's start or resume procedure works.

PR #610 (2025-01-10): Fix IC bike SE showing only zeroes
The IC bike SE will stop sending notifications if an unsupported Fitness Machine Control Point procedure is used. To prevent it, read the machine fitness features to determine if the Elapsed Time Supported and the Remaining Time Supported bits are set in order to determine if the "Start or Resume" can be used.

Also, reduce the connection parameters to 100 instead of 120 to be able to get the notifications which are sent every 500 ms.

This fixes #605
closed 2025-01-10 by doudar