doudar/SmartSpin2k #612: Resets during ERG training with power changes

serbasan, 2025-01-10:
I've noticed an issue where I experience resets during ERG mode training sessions. This usually happens when there’s a significant power change, for example, going from 115W to 245W.

In a 1-hour session, this can occur between 0 to 3 times. Today, I was finally able to capture a log of the issue since it happened while my computer was on.

The reset occurred around line 12000 in the attached log.
Additionally, I noticed that the PowerTable was emptied at line 11278, which might be related.
This issue has been happening for weeks, but I hadn’t been able to capture it until now.

Attachments:

Log file with the reset moment.
Thank you in advance for your time and help in investigating this issue.

[Custom UDP Receiver (Port 10000).csv](https://github.com/user-attachments/files/18381807/Custom.UDP.Receiver.Port.10000.csv)

[image]

doudar, 2025-01-12:
I ride ERG a lot and haven't been seeing any reboots. I looked at your log, there's a reboot. Looking at this, It's having a hard time creating anything much for the power table. That still shouldn't cause a reboot, but it might be what's exacerbating it. My guess is what's actually causing a reboot is an insane number being returned from one of the lookup functions. I'll look into adding more protections there. 

In the meantime, I'd recommend you (Clear active powertable using the app first) give homing a try on your bike. Select the calibrate button on any training app, start pedaling and let the knob go to it's limits. Once it starts moving, you can stop pedaling and let it do it's thing. Once you enable homing, then on every power cycle, the SmartSpin2k will seek to the lowest resistance position on your bike as soon as you start pedaling. Pressing a shifter before or during pedaling will abort the procedure. You can deactivate homing by doing "clear active powertable" using the app. 

Edit: Also, I just remembered that you have a bit of a custom bike setup that requires very small movements to change power. That's why the power table is having a hard time in your case, because the numbers are similar. You use very low shift steps, correct? Try increasing the number here (to 8 or maybe even 16) and let me know if it helps. https://github.com/doudar/SmartSpin2k/blob/705b7d203deda7e61de0b8e416747a2f415efc47/src/Main.cpp#L559

serbasan, 2025-01-12:
Hi Anthony,

Thank you for suggesting increasing the Steps to 16. I've tested this configuration and attached two logs for your review:

Calibration log:
I captured this log during calibration from Zwift after clearing the PowerTable. However, I don’t see the behavior you mentioned about reaching the lower and upper limits during calibration. The values only move slightly at times. Similarly, nothing unusual happens when I start SS2K and begin pedaling.

ERG test log:
This log corresponds to a test training session in ERG mode. Could you check if the PowerTable now looks appropriate with 16 Steps?

Thank you for your help and support!

Attachments:

Calibration log
ERG test log
[Calibracion 16steps.csv](https://github.com/user-attachments/files/18388971/Calibracion.16steps.csv)
[erg 16steps.csv](https://github.com/user-attachments/files/18388972/erg.16steps.csv)

serbasan, 2025-01-12:
Hi Anthony,

Following up on the issue with homing that I mentioned earlier, I noticed in the documentation that the SS2K version 2 uses a TMC2225 driver, while version 3 seems to include a TMC2209 or a similar driver.

Could this difference be the reason why the homing function isn’t available on the v2?
If so, would it be possible to replace the TMC2225 with a TMC2209 on the v2 to enable this feature?

Thanks for your help and for the great work on this project.

Best regards,

doudar, 2025-01-12:
Okay, yes the 2225 doesn't support sensorless homing. I didn't realize you were using a version 2. In addition to using a 2209, You would also need to look at boards.h and settings.h and add the version resistor and make all of your connections the same as version 3, as some of the pins changed.

doudar, 2025-01-14:
It would enable homing, otherwise, there's no change. I'm not sure how homing would work on your bike either because of the constant pull - it might false trigger the high limit prematurely, but you might be able to increase homing force enough to compensate for that. I'm glad that you were able to get this going!

closed 2025-01-24 by doudar