doudar/SmartSpin2k #622: [BUG] Motor went past lowest bike resistance

madxime, 2025-02-03:
### Description

I was doing a workout in Rouvy where the goal was to keep a 85 RPM cadence with 120 RPM 15 seconds bursts. When I reached the first burst, the bike resistance dropped almost instantly when I started spinning faster, and the motor kept turning past the minimum resistance of the bike's knob which broke the internal spur gear (i.e. this one: https://github.com/doudar/SmartSpin2k/blob/develop/Hardware/Common%20Assets/Gears/Spur%20gear%201M%2011T.STL) 

### Steps to reproduce

I don't know

**Expected behavior**
The resistance shouldn't decrease or increase past some failsafe value

**Actual behavior**
The resistance went past the bike's minimum resistance value

### Additional Information
I was using an IC Bike SE with smartspin2k version [25.1.10](https://github.com/doudar/SmartSpin2k/releases/tag/25.1.10). I had kept all the default tuning values. If that can be useful, I noticed that my bike's minimum reported resistance is 1 when the knob is set to the lowest value, not 0. 

Is there something I should've tuned that would've prevented this issue from happening?

jekkos, 2025-04-02:
I had similar issue when using erg, of has been quite unreliable.for me. Is there any way to calibrate the trainer not to go below a certain value?

doudar, 2025-04-02:
If you have a bike with defined mechanical stops, (and you're using an R3 SS2k), you can "calibrate trainer" in any app and the SS2k will find the travel limits of the knob and respect them. You'll probably want to use the SmartSpin2k companion app and "clear existing power table" first. 

There's a lot of work going on in the testing branch to improve related things, we're in code cleanup and bug fixing stage on the testing branch and it's not fully stable yet.