doudar/SmartSpin2k #628: [BUG] (short description)Periodic Data Drops with Tempo Power Meter

WayneThayer0, 2025-02-15:
### Description

When working with a Tempo Power Meter, data dropouts occur every 5 minutes (to the second). In the companion app, this is seen by the Cadence and Power staying constant at the last seen value. If the Tempo Power data is forwarded by SS2k to Zwift, during the dropouts the Cadence and Power both go to zero (and if HR is also passed through SS2k, it also goes to zero). With a Schwinn IC4 as the source for Cadence and Power, there are no dropouts. If Zwift is set to use SS2k/IC4 for Resistance Control, but take Power and Cadence from the Tempo Power Meter, then there are no dropouts. 

### Steps to reproduce

1. Boot up SS2k and allow to "Home"
2. Using the SS2k Companion App/Settings, verify that the Tempo Power Meter is selected as well as the HR sensor
3. Launch Zwift and configure for Power, Cadence, HR, and Resistance all passing through SS2k
4. Select a Zwift ride of over 5 minutes
5. Watch for the data to go to zero, or analyze the ".fit" file

Alternatively, use the SS2k Companion App to run a workout "Andrew's Mix" and watch for the Data to go absolutely constant (real time or analyze ".fit" file).

**Expected behavior**
Expected no dropouts.

**Actual behavior**
Dropouts

### Additional Information
Probably the Zwift workaround of using SS2k for the Resistance control but using the Tempo Power Meter for Power will cause issues because the SS2k will have a different power reading than Zwift.

doudar, 2025-09-16:
Testing #674

doudar, 2025-09-16:
[firmware.zip](https://github.com/user-attachments/files/22366827/firmware.zip)
@WayneThayer0 I might have identified the issue. If you could extract this firmware and test it, I'd appreciate it!

WayneThayer0, 2025-09-16:
Should be able to do that today. Thanks!

doudar, 2025-09-16:
So happy this works! Thanks for testing!

WayneThayer0, 2025-09-16:
> That's because the power table version changed between firmware versions.
> Do another calibration in any app and you'll be good to go.

What app has calibration? I am running the SmartSpin2K app on android.

doudar, 2025-09-16:
Almost any of them. If you're using the SmartSpin2k companion app for workouts, go to the workout screen, then click the ... in the upper right corner, and you'll find calibrate in that menu.

WayneThayer0, 2025-09-16:
Ok. All set now. Thanks!

doudar, 2025-10-19:
#fixed by #674

closed 2025-10-19 by doudar