doudar/SmartSpin2k #630: [BUG] Invalid BLE value length (GTBike V crash)

claabs, 2025-03-02:
### Description

When connecting to [GTBike V ](https://www.gtbikev.com) via BLE, the connection will make a few readings and then crash GTAV. Digging into the  GTBike V logs, this error message related to BLE appears:

```System.ArgumentException: Destination array is not long enough to copy all the items in the collection. Check array index and length.
<...>
at Ble.Devices.Services.CPService.<Start>d__46.MoveNext()
```

Posting in the [GTBike V support forum](https://www.facebook.com/groups/1089053124812221/posts/2302052786845576/), the developer concluded this:

> Looks like your trainer is sending a non compliant value in the Cycling Power Feature characteristic. The BLE standard specifies that as a 2 byte value, but your trainer is sending a 1 byte value. When trying to convert that into a ushort, there is not enough data in the received array to make the conversion and it fails. I could make this a bit more robust, but it's very difficult to support hardware that does not comply with the standards.

The ask here is to **ensure that SS2K is following the BLE spec for these values.**

### Steps to reproduce

1. Install GTAV and GTBikeV 
2. Install the latest ScriptHookV and ScriptHookVDotnet
3. Have SmartSpin2k ready for BLE connection
4. Launch GTAV and enable the mod. The game will crash to desktop a few seconds later.

**Expected behavior**
The game should not crash and work, as it is claimed to be compatible [in the docs](https://docs.smartspin2k.com/compatibility.html#supported-training-programs).

**Actual behavior**
The game crashes.

### Additional Information
* Logs and config for my GTBikeV setup: https://gist.github.com/claabs/25f458cdc59a24d6b41ea4bbee0c6d68

matthewshirley, 2025-03-13:
I also had this issue today and thought it was my game. 🤯 

I looked at this quickly, and it is supposed to be 4 bytes, not 1 or 2 bytes [1] [2]. This [line](https://github.com/doudar/SmartSpin2k/blob/develop/src/BLE_Cycling_Power_Service.cpp#L19C3-L19C108) of the `BLE_Cycling_Power_Service` should be updated to be 4 bytes:

```
byte cpFeature[1]                     = {0b001100};  // crank information & wheel revolution data present
```

Bits 21 to 32 are reserved, so we can set these as 0. I also don't believe we need to flag bits 9 to 20. I'll try a fix tonight :)

[1] [GATT Specification](https://btprodspecificationrefs.blob.core.windows.net/gatt-specification-supplement/GATT_Specification_Supplement.pdf) (Section 3.65)
[2] [org.bluetooth.characteristic.cycling_power_feature.xml](https://btprodspecificationrefs.blob.core.windows.net/gatt-specification-supplement/GATT_Specification_Supplement.pdf)
[3] [Cycling Power Service](https://www.bluetooth.com/specifications/specs/cycling-power-service-1-1/)

doudar, 2025-03-13:
Awesome! Thanks for your help on this.

doudar, 2025-03-13:
Also, since that was written, we should be able to now build the flags field using the pre-defined flags struct in BLE_Definitions.h

matthewshirley, 2025-03-13:
I connected to GTBikeV with no crashes after updating the characteristics to 4 bytes. 🎉 

@doudar Thank you for highlighting that example. I'm a big fan of no magic numbers!

PR #632 (2025-03-13): Correct Cycling Power And CSC Feature Characteristics 

It was reported that GTBikeV crashes when a SmartSpin2k device connects as the length of the Cycling Power feature characteristic is only 1-byte. As noted in the following [comment](https://github.com/doudar/SmartSpin2k/issues/630#issuecomment-2719457987), this characteristic should be 4-bytes.

Therefore, this PR updates the Cycling Power feature characteristic to be 4 bytes, with the 2nd and 3rd bit flagged. I additionally updated the Cycling Speed Cadence, which is supposed to be 2 bytes. However, the last byte is reserved, so I doubt  [... cut]
closed 2025-03-13 by doudar